bdring/Grbl_Esp32 #126: Any attempt to move axis after Z-Probing leads to a crash

vogtitec, 2019-04-19:
Procedure: Z-Probing by G38.2 Z-10 F10. Works fine. If you then try to move any axis (e.g. with jog), the ESP crashes in stepper.cpp with the following data :

> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40081124  PS      : 0x00060031  A0      : 0x40081d7c  A1      : 0x3ffbe900
A2      : 0x00000000  A3      : 0x3ff5f000  A4      : 0x3ffc1c78  A5      : 0x00000000
A6      : 0x0477e351  A7      : 0x3ffc1bb4  A8      : 0x80081120  A9      : 0x3ffbe8e0
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x800db6e5  A13     : 0x3ffb1db0
A14     : 0x00000002  A15     : 0x000000fc  SAR     : 0x00000005  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000011  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff
Core 1 was running in ISR context:
EPC1    : 0x40081124  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40082edd

> ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

> Backtrace: 0x40081124:0x3ffbe900 0x40081d79:0x3ffbe930 0x400dba95:0x3ffb1df0 0x400d276f:0x3ffb1f90 0x400f4c09:0x3ffb1fb0 0x4008f451:0x3ffb1fd0

> Rebooting...

The exception decoder tells:

> PC: 0x40081124: onStepperDriverTimer(void*) at C:\Users\cvogt\AppData\Local\Temp\arduino_build_125167\sketch\stepper.cpp line 265
EXCVADDR: 0x00000011

> Decoding stack results
0x40081124: onStepperDriverTimer(void*) at C:\Users\cvogt\AppData\Local\Temp\arduino_build_125167\sketch\stepper.cpp line 265
0x400dba95: protocol_exec_rt_system() at C:\Users\cvogt\AppData\Local\Temp\arduino_build_125167\sketch\protocol.cpp line 428
0x400d276f: loop() at Y:\Arduino\ESP32_GRBL_WebUI_neu_180419\Grbl_Esp32-master\Grbl_Esp32/Grbl_Esp32.ino line 141
0x400f4c09: loopTask(void*) at D:\Benutzerdateien\cvogt\Downloads\arduino-1.8.5\hardware\espressif\esp32\cores\esp32\main.cpp line 25
0x4008f451: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

Any idea how to solve this would greatly be appreciated!!
Chris

bdring, 2019-04-19:
I am able to recreate that problem. Here is what I see. Some jog types work and some do not. Normal moves appear to work and then all jogs work after that. 

Here is a typical probe sequence that works

G38.2 Z-40 F100   ; probe down 40 at speed 100

[PRB:-299.000,-1.000,-8.095:1]  ; grbl response of successful probe

G10 L2 P0 Z-28.095  ; set Z offset in current coordinate system to the returned Z value - touch plate thickness (20)

Either of these commands below work and all subsequent jogs are OK

G0 Z20 ; move to the probe thickness
$J=G53 G21 Z-1 F200   ;jog to Z-1mm in machine coordinates at speed 200

This problem might take a while to track down, but I think it might be due to the machine left in a problematic mode. doing a normal move cleans things up.

If you want to discuss short term fixes via Slack, let me know and I'll send an invite.

bdring, 2019-04-20:
@vogtitec  BTW: What sender are you using?

vogtitec, 2019-04-20:
@bdring Thanks for the first solution attempts. I first encountered the behaviour with candle 1.1.7, but I was able to reproduce it by entering the probing command manually in the WebUI and then jogging via Web.
Yes, Slack will be welcome, thanks!

vogtitec, 2019-04-20:
Ok, so there is a simple workaround for the problem! I changed the standard probe command in candle by adding a simple G91G0Z10, and now it works! Thanks a lot!

bdring, 2019-04-20:
Great.

I added [Candle](https://github.com/bdring/Grbl_Esp32/wiki/Controlling-Grbl_ESP32) to this wiki page.

109JB, 2019-04-22:
I'm not near my machine so can't check myself but is it possible to use a G91G1X0 command, which would not actually move anything,  to "reset" things after the probing?

bdring, 2019-04-23:
That does not work. It appears you need to have some movement

bdring, 2019-04-29:
I think I found the problem. Thanks to the [ESP Exception Decoder](https://github.com/me-no-dev/EspExceptionDecoder), I found the offending line of code. 

The stepper interrupt was crashing while trying to compensate the spindle speed vs feed rate. This is a feature used by lasers and does not apply during jogging. I just placed a test for joging around the code.

I'll release it soon to the devt branch. Then merge to master soon.

closed 2019-04-29 by bdring