bdring/Grbl_Esp32 #127: Proposal for software debouncing limit switches

vogtitec, 2019-04-24:
Sorry for posting this as an issue, but I'm very new to git and I didn't know where else to publish my idea:
I found a simple solution for debouncing and eliminate interferences on the limit switches by adding some subsequent extra reads of the inputs in the ISR before firing an alarm.
Since delays can't be used in an ISR, I do an adjustable number of reads, in my case 3 worked fine.
I added following declarations in the top of "grbl_limits.cpp":

```C
volatile byte readresult=0;
#define DEBOUNCEREADS 3
```

Then, at the beginning of `void IRAM_ATTR isr_limit_switches()`, I do subsequent readings of the limit inputs:

```C
readresult=0;
for (byte N=0; N<DEBOUNCEREADS; N++){
    if ((digitalRead(X_LIMIT_PIN)==HIGH) || (digitalRead(Y_LIMIT_PIN)==HIGH) || (digitalRead(Z_LIMIT_PIN)==HIGH)) {readresult++;}
}
```

Only if ALL subsequent reads show a triggered input, the rest of the ISR shall be executed. So I wrapped the whole rest of the code in this condition:

```C
if (readcount==DEBOUNCEREADS){
    if ( ( sys.state != STATE_ALARM) & (bit_isfalse(sys.state, STATE_HOMING)) ) {
        if (!(sys_rt_exec_alarm)) {
            [...]
        }
    }
}
```

Before my modification, I had some very annoying alarm triggers by interferences from my HF spindle, despite of having applied all hardware means of debouncing (capacitors and strong pullups). The modification eliminiated all theses interferences!
**BUT BEWARE!! :
My modified code doesn't take into account any inverting settings for the limit pins!** 
As I'm not so deep in the grbl code, I just didn't know how to manage this...
Perhaps someone could pick up the idea and correct the code?

bdring, 2019-04-24:
I added this to the [roadmap](https://github.com/bdring/Grbl_Esp32/wiki/Development-Roadmap)

AVR Grbl uses a timer to do this controlled by #define ENABLE_SOFTWARE_DEBOUNCE in config.h

I am worried that 3 iterations would run so fast that it might not work for all people, but increasing that might affect performance. We also have the option of using a task instead of a timer.

bdring, 2019-04-26:
I created a new [branch](https://github.com/bdring/Grbl_Esp32/tree/Limit_Debounce) to test a different type of limit switch debouncing.  Original Grbl uses a timer (watchdog) to recheck the switches after about 32ms. I tried the same thing using a delayed RTOS task. If you (@vogtitec ) could try it for me, that would be helpful.

I think a task based system is more flexible and less of an impact on Grbl. If it does not work, I could try a hybrid system where a faster task counts consecutive reads. 

Thanks

vogtitec, 2019-04-28:
@bdring Today I tested your limit switch debouncing branch. At first glance, it seems to work very fine!! I couldn't encounter any problems testing with two nc-files, and I had no interferences on the inputs!

By the way a short question: the web interface shows buttons for "spindle" and "flood". Both are set up in my configuration and can be controlled via M-commands, but the web buttons don't do anything... ???

luc-github, 2019-04-28:
@vogtitec indeed there is a bug in sending realtime command in webUI  - I am looking at it now
seems command sending `String.fromCharCode(0xA1)` send actually 0xC2 and 0xA1, and command sending `String.fromCharCode(0xA0)` send ...nothing, they are not printable char so the behavior is may be  not accurate, I will find a solution ASAP

bdring, 2019-04-28:
@vogtitec Thanks for testing that. I'll put that in the devt branch merge it with master soon.

bdring, 2019-04-28:
@vogtitec I am working with @luc-github on the spindle button and he fixed an issue, but the way that button works is not very clear.

If you are running a job with a spindle on and you do a feed hold, that button allows you to toggle the spindle off and then back on. If you resume the job, the spindle will come back on regardless of where you left the toggle state. The Mist and Flood buttons work in a similar fashion.

This is an advanced feature of Grbl that I really did not fully understand until now. We might change that panel title to "Overrides" to help make that clearer. 

A spindle control panel doing what you want would require a speed value to  work. M3 without an S value is a problem.

closed 2019-04-28 by bdring