bdring/Grbl_Esp32 #128: Connecting to a CO2 laser PSU

kjjordans, 2019-04-25:
Hi,
I have a Chinese Laser cutter and I am planning to use GRBL_ESP32 controller for it and i have found it very hard to find any documentation on how to connect the controller to the laser PSU. I know that the parts vary slightly from machine to machine so i'm documenting my experience on hackaday.io to try to help out the community. I would really appreciate it if you could take a look at my wiring and give me some advice. this is what i think:
1. connect the laser PWM pin to the TH pin through a 3.3 to 5v logic level shifter. (i think it may work without the shifter).
2. I'm not sure if i need to have a laser enable pin.
3. i think that i can still use the test button that is connected to the TL pin
4. I plan to remove the stock controller completely and i am not sure what power level the laser will be set to when I press the test button
5. i have read all over the internet that i should install an ammeter between the PSU and the earth wire from the laser so i will probably do that but i don't know if it will affect the laser operation (any ideas?)
6. after i have installed the ammeter, i will try to set the max current to 25mA as suggested by the laser supplier. (it also says 25mA max on the PSU so I might not have to change anything.)
Sorry for such a long question. I hope that this is the right place to ask. If you Think that there is a better please to ask, let me know. 

check out this for more details and photos :) 
[https://hackaday.io/project/165206-upgrade-chinese-laser-controller/log/162465-laser-psu](url)

gflaser-au, 2019-04-25:
I have grbl32 working on my 100w co2 laser.
Works well.

5, ammeter goes in-line in the negative psu output wire.  This is the only safe option.
If you put it in the other high voltage line the entire ammeter will be at very high voltage. Not safe.

1, I used a high speed opto-coupler to do the level shifting.  A comms opto. The opto will help with interference, but is not necessary for isolation (normally).

2, I wired the psu enable line to an “enable” switch that I can control manually.  It’s a safety feature ;)

3, you won’t need it once you are past alignment activities.  I just made a manual wire harness with a potentiometer to set power and a push button sw to fire. Fully manual. Fully disconnected from controller board.

6, your psu will likely output more than 25ma. Start with low power settings and work up.  It helps to use an oscilloscope before hitting the laser ;) get a feel for the S value that will set 10% pwm.  This is about where the laser will begin operation (min power). Remember the psu is controlled by voltage not the pwm (my experience) and you will need to set the pwm frequency to be 20khz, or more. Also remember to check if the level shifter inverts the signal.

Last advice.  Interference can be a challenge.  You have a 30-40 kilovolt arc in the laser tube.  It’s like a massive wideband radio transmitter with quite some power. It is advisable to use shielded cable and be very careful with where and how everything is grounded. Keep the laser PSU physically separated and shielded from the controller pcb.

There are some grbl build options you will need to setup too....  happy to help once you get to that point.

bdring, 2019-04-25:
I would love to help. I would be happy to design some custom hardware for this because so many people have been asking for it. I could provide one for free for testing.

 - What is the donor laser you are starting with?
    - Does it have stepper drivers? They might need 5V with enough power to light an opto.
    - Does it have limit switches? How are they wired?
    - Water flow meter?
    - Cover Switch?

The WP line is a safety interlock. I recommend running it through a few things for safety, like the flow meter, cover switch and maybe a key lock. We could have Grbl monitor that line, so the job stays in sync with the interlock.

Some 5V level shifters/isolators have an enable line that is a nice way to add a little bit of safety from the controller going rogue.

You might be able to use a pot as a settable divider to limit voltage into the p/s. That might be able to work with the test fire button. It is nice to align the laser with low power pulses.

I could setup a Slack Channel if you want to collaborate easier.

kjjordans, 2019-04-26:
Hi, thanks for the great advice. I would be more than happy to collaborate with you. 
I have been taking some photos of the existing hardware and I will draw up some schematics and upload to the hackaday project page. I haven't used slack before but I'd be happy to give it a try.

kjjordans, 2019-04-26:
Hi,
I added some info on the limit switches and motor control. 

https://hackaday.io/project/165206-upgrade-chinese-laser-controller/log/162520-stepper-limit-switch-daughter-board

bdring, 2019-04-27:
It is hard to tell what is going on with that. Can you post more pictures and info. Also put those photos into the context of where they mount and the wiring.

kjjordans, 2019-04-28:
I have updated the log with a full wiring diagram. 
https://hackaday.io/project/165206-upgrade-chinese-laser-controller/log/162617-the-wiring-diagram
I have also uploaded one with a white background

bdring, 2019-04-28:
That is helpful. What parts do you want to keep?

What size are the stepper motors?

kjjordans, 2019-04-29:
hi 
the steppers are ordinary NEMA17 motors. I want to replace the controller and laser power board. 
I want to add door interlocks and a high temperature cut off for the cooling system. I have some waterproof ds18b20 which I could use.

bdring, 2019-04-29:
Do you want to monitor temperature and report it? I think a more robust solution would be to use a temperature switch. This would allow it to save the laser even if the controller has an issue.

kjjordans, 2019-04-29:
I'd be happy with a switch. it would be nice to know what has caused the fault. ie. flow, temp or open door. 
do you know of any temp. switches, ideally something I can put in line with the water outlet hose?

kjjordans, 2019-04-29:
it would also be nice to know what the water temperature is so that I know to add more ice before it over heats. 
this probably dosent need to be on the controller board. I could get a stand alone temp sensor and display.

gflaser-au, 2019-04-30:
I used a PC water cooling radiator that fits two 120mm high speed fans in my 100w laser.
I have a thermometer in the header tank and even when cutting at 28-30 degrees C ambient, and reasonable power, I don’t see crazy temps. High 30s at most.
Personally, I wouldn’t bother with the extra complication of a temp sensor - the water flow switch connected to the laser psu is most important.

What I did do is connect the “flood” output to a mosfet to switch an air solenoid to control the nozzle air. This rocks.  It saves me from either forgetting to switch the air on or off. I could also use this to switch the cooling fans too but haven’t bothered.  I manually control the fans / pump / laser enable using switches on the machine’s control panel.

kjjordans, 2019-04-30:
hi.
I've read on the internet that you should keep the temperature of the coolant below about 25 degrees although, I have never seen a proper datasheet for one of these lasers.
my water pump and air assist are connected to outlets on the back of the machine which are always on. they are not that noisy so I will probably leave it as it is.

kjjordans, 2019-04-30:
I just saw a digital temp. controller on eBay for less than $5 with a digital display. maybe I could connect it to a normally open relay wired in series with the interlock switches. it would be a nice meet solution which would remain independent of the controller.

gflaser-au, 2019-04-30:
My Reci laser spec is 40degrees.
I have also read that too cold can also be an issue.... if you’re using ice, be careful with that. I don’t have to worry about that where I live :)

http://www.recilaser.com/en/index.php?m=&c=Index&a=show&catid=7&id=66

kjjordans, 2019-05-18:
Hi again.
I have been thinking about driving the 5v pwm. I saw this circuit which I think is a good start. What do you think? https://redmine.laoslaser.org/boards/2/topics/839
I think this circuit will invert the pwm. Should I use an inverter or is it easier to map the pwm from the esp32 100% duty for 0% laser power.
I also have a logic level shifter that uses mosfet like this one https://www.sparkfun.com/products/12009
But I’m worried about the switching speed. The cuttent controller uses 22kHz
I would really appreciate your help.

brakthehun, 2019-06-11:
To connect to that power supply - you need to connect water protect to ground. connect the ground of the ESP32 to the laser PSU ground. Then you connect your trigger pin to TTL high.
This is all you need to run this type of laser PSU.

If you want,, you can use an NPN transistor,, and put a 2K resistor from Laser PSU 5V to Laser PSU TTL low. then output pin on the ESP32 to the NPN.

closed 2019-11-05 by bdring