bdring/Grbl_Esp32 #129: Issue to control Grbl_Esp32 with software candle

sirias52, 2019-04-27:
Serial Conexion response
[image]
But control GRBL_ESP32 it not possible.

gflaser-au, 2019-04-27:
Did you use “unlock” $X command before others?

Try these commands listed here.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands

sirias52, 2019-04-28:
if you use that option but it does not work... 
I'm trying with these programs (https://github.com/bdring/Grbl_Esp32/wiki/Controlling-Grbl_ESP32), 
but I can not make it work

bdring, 2019-04-28:
I just downloaded Candle and it works fine for me.

bdring, 2019-04-30:
All of the programs on that wiki page have been tested at some time.

bdring, 2019-05-02:
I am going to close this issue because it sounds like general serial port issue. I would be happy to help work out the issue over Slack. Request an invite.

closed 2019-05-02 by bdring

sirias52, 2019-05-10:
Position Type | 1 | Enable MPos:. Disable WPos:

esp32 GRBL default position type is 2