bdring/Grbl_Esp32 #133: Android bluetooth serial  only read MPOS and WPOS in mm

mac7988, 2019-04-28:
Hi,

for some reason it only reports in mm. I tried to change $13=1 but no effect.

bdring, 2019-04-29:
It works fine for me. Have you tried it in a simple terminal in both USB/Serial and Bluetooth?

Here is a screen shot over Bluetooth with my phone.

[image]

mac7988, 2019-04-29:
Hi Bdring,

I am using grbl controller by can silicone and it seems no matter what I do it won't change it.

Maybe it's the application.

Thanks for trying to help me!
Cheers!

bdring, 2019-04-29:
I tested with Grbl Controller as well. It was fine. It was easier to screen  capture on terminal

bdring, 2019-04-29:
Maybe post the question at that repo too.

mac7988, 2019-04-29:
I think I am on the old version of  your code. I checked the same thing and it seems to be with the decimal  points which was previously reported.I will update to the newest code and report back.

thanks again.

closed 2019-04-29 by bdring