bdring/Grbl_Esp32 #134: New code bluetooth wont turn on

mac7988, 2019-04-29:
just updated to the latest code and bluetooth wont turn on

$I="NAME" 
then I reset but no bluetooth.

bdring, 2019-04-29:
The default system now compiles with both bluetooth and wifi enabled, but you can only use one at a time.

The $I method was from the early days of Grbl_ESP32. As we added a lot of new features and parameters, we needed a universal command system to deal with them.

We now use the [ESP...] commands.
https://github.com/bdring/Grbl_Esp32/blob/master/doc/Commands.txt

You would need to set the Bluetooth name with ...
[ESP140]< Bluetooth name> pwd=\<admin password\>

If you want your BT name to be frodo, you would send
[ESP140]frodo
The default compile does not require the password part. If you have a password you would send
[ESP140]frodo pwd=admin   (assumes admin is your password)

You also need to change your radio state to BT with...
[ESP110]BT

mac7988, 2019-04-29:
Ok! This explains everything. I am clearly on the wrong versions. 

Thanks alot!

closed 2019-04-29 by mac7988