bdring/Grbl_Esp32 #138: Program too big for nodemcu-32s?

AlmightyFrog, 2019-05-04:
I was about to give grbl_esp32 a first try on a real machine. I've already wired up a board since some time, using there `UI: 2.1b7 / FW: 1.1f`.

Now I just fetched current master trying to update my board, but it looks like size is not fitting anymore:
```
Processing nodemcu-32s (platform: espressif32; board: nodemcu-32s; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Converting Grbl_Esp32.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FS> 1.0
|-- <EEPROM> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <WiFiClientSecure> 1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <ESP32SSPD> 1.0
|   |-- <WiFi> 1.0
|-- <WebSockets> 2.1.2
|   |-- <SPI> 1.0
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <SPI> 1.0
|-- <WebServer> 1.0
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|-- <BluetoothSerial> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <SD(esp32)> 1.0.5
|   |-- <FS> 1.0
|   |-- <SPI> 1.0

(...)

Linking .pioenvs/nodemcu-32s/firmware.elf
Retrieving maximum program size .pioenvs/nodemcu-32s/firmware.elf
Building .pioenvs/nodemcu-32s/firmware.bin
Checking size .pioenvs/nodemcu-32s/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
Error: The program size (1791566 bytes) is greater than maximum allowed (1310720 bytes)
DATA:    [==        ]  21.0% (used 68748 bytes from 327680 bytes)
PROGRAM: [==========]  136.7% (used 1791566 bytes from 1310720 bytes)
*** [checkprogsize] Explicit exit, status 1
esptool.py v2.6
```

Looking at the numbers, even if changing partitioning i guess it would still not fit. What am I missing or doing wrong?

bdring, 2019-05-04:
Check your partitian.  Read the wiki

luc-github, 2019-05-04:
read wiki : https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware
[image]

luc-github, 2019-05-04:
Bluetooth use a lot of flash - if you do not plan to use but want bigger SPIFFS you can disable SPIFFS in config.h

AlmightyFrog, 2019-05-04:
Thanks for fast answer, changing partition indeed did help. Here results using [min_spiffs.csv](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/min_spiffs.csv):
```
DATA:    [==        ]  21.0% (used 68748 bytes from 327680 bytes)
PROGRAM: [========= ]  91.1% (used 1791614 bytes from 1966080 bytes)
```
I'll verify later that day if it is working fine and if so create a PR for it.

luc-github, 2019-05-04:
Sorry, a PR for what ? What is the issue using Min SPIFFS partition?

AlmightyFrog, 2019-05-05:
@luc-github there is everything fine with code. It is just like when using platformio instead of arduino UI you need to define the "non default" partition schema.

Looks like maybe no one besides me is using platformio for Grbl_ESP32 besides me.

Please note, that of cause to get that partition schema updated on ECU, you need to to a target upload over usb in pio as OTA is not working for that.

luc-github, 2019-05-05:
Understood, thank you for explaination

bdring, 2019-05-05:
Thanks @AlmightyFrog 

This was added to pr #137

closed 2019-05-05 by bdring