bdring/Grbl_Esp32 #140: invert spindle enable pin

jschoch, 2019-05-05:
I'm using board 2.1 and trying to get a WS55-220S brushless spindle controller working.  This pesky device requires enable low...   I compiled and uploaded a changed config.h but it does not seem to be working.  I had a few questions.

PWM:

PWM control doesn't work even if I jumper enable to GND.  the manual says 

``` 
pulse signalcontroller to provide customers with the motor speed is proportional to the terminal by "PG", the pulse amplitude of 5V .
= Rotational speed of the pulse motor pole number frequency ÷ × 60 (seconds);
```

I can see 5v on the PWM input to the controller when I max out the speed after installing a level shifter.  The voltage seems to be proportional and correct but I do not have a scope.  I've tried to set the max speed to 100 and to 30000 which I read changes the PWM duty cycle.  None of them seem to result in the controller registering the pwm speed.

Enable Pins

1. is board 2.1 GPIO 32 the correct pin for spindle_enable_pin?
2. the comments say that if you want to enable INVERT_SPINDLE_ENABLE_PIN and use VARIABLE_SPINDLE you also need to also enable USE_SPINDLE_DIR_AS_ENABLE_PIN.   I've looked at the pins on both spindle_enable and spindle_dir and oddly it measures 4.5 volts!  I also thought that grbl mega had removed this limitation.  I'm not sure how this version works or what it was derived from.

Bottom line is I'd like to have a LOW enable pin somewhere.  I could do without PWM but I'd like the spindle to stop on a fault.

terjeio, 2019-05-05:
Did you try speed control with the supplied potmeter? The controller requires 0-10V DC input for speed control, IIRC PWM did not work for me. I added a level shifter followed by a low-pass filter to get it working.

jschoch, 2019-05-05:
the pot works.  the SV is for the 0-10v signal.  the manual says it should work in the PG input with 5v.  I had a setup working with this spindle before but I may have used a relay to enable the spindle which is what i'll do short term.

i am still perplexed why i don't get any change on either spindle_dir or spindle_enable pins with VARIABLE_SPINDLE, INVERT_SPINDLE, and USE_SPINDLE_DIR_AS_ENABLE_PIN all enabled.

jschoch, 2019-05-05:
I was able to invert the PWM signal with a 2n2222 using this circuit:  https://forum.arduino.cc/index.php?topic=249116.0

this works and if the controller is reset it should not hickup and go low.

jschoch, 2019-05-07:
quick update:  I was able to get this working.  I had quickly scanned for the SPINDLE_ENABEL_PIN definition but it was wrapped in a config block.  it does not appear the pin is inabled by default.  I enabled both SPINDLE_ENABLE_PIN and SPINDLE_DIR_PIN in config.h and commented out ENABLE_SD_CARD.  I was not sure which pin would respond to M3/m5.  it turns out that the SPINDLE_ENABLE_PIN was correctly inverted and responding to the M3/m5 commands.  I have not yet tested power up behavior and it may blip on the spindle on startup (this is what an arduino would do).

So, net net is you need all of VARIABLE_SPINDLE, INVERT_SPINDLE, and USE_SPINDLE_DIR_AS_ENABLE_PIN, SPINDLE_ENABLE_OFF_WITH_ZERO_SPEED enabled and you also need to map at least  SPINDLE_ENABLE_PIN (v2.1 board i used GPIO_NUM_18).

Hope this helps someone.  PWM still not working for this crappy brushless controller :(

closed 2019-05-07 by jschoch

terjeio, 2019-05-07:
Crappy or not I got it working in my mini lathe... I had to add closed loop RPM control (PID) as the motor I am using did not live up to the hype. Here is the schematic for the level-shifter I made:

[image]

IIRC a transistor (for signal inversion) is needed to drive the opto-coupler to get it working correctly with the ESP32 - I am using a MSP432 where I can invert the PWM output via a setting.

May I ask what motor you are using?

jschoch, 2019-05-07:
The motor/spindle is labeled the same as the controller and came with the controller.  It has two small deep groove ball bearings and a fairly wobbly ER16 collet mounted to the shaft.  There is significant axial play which makes it not suitable for much other than PCB milling which is what it is dedicated to now and until it dies.   I get about 7krpm @27v out of it with the paired controller.

I'd love to find a battle tested high rpm small spindle that uses angular contact bearings.  The wegstr spindle looks great but is pretty expensive.  The wolf engineering spindle is not available to purchase anymore from my searching.  I looked into the Sable-2015 but it uses regular ball bearings.