bdring/Grbl_Esp32 #146: bCNC Issues

bdring, 2019-05-13:
Someone using the [bCNC](https://github.com/vlachoudis/bCNC) reported connection issues. I was able to confirm that and found (2) issues.

**First**

It will not connect if the $10=x reporting mask is such that x is not an odd number. The default is 2.

Does anyone have a problem changing that to 1 or 3? 

**Second**

It asks for the gcode modal status with $G after some commands are sent. Grbl_ESP32 is sending...

[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F:0 S0.000]

It should have been…

[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0.000]

There was a colon after the F.

bCNC locks up without reporting an error when that happens.

I will fix that with a pull request soon.

AbySet, 2019-05-21:
You guys using bluetooth or wifi to connect ?

bdring, 2019-05-21:
Both...Bluetooth is better for streaming and wifi is better for the WebUI and SD card

closed 2019-06-07 by bdring

CNCmaisdeMILL, 2019-11-02:
a friend is having this problem, i hope for a future fix because bcnc is the best sender.

bdring, 2019-11-02:
The problems mentioned on this Closed Issue have fixed.  It works fine for me now.  Please open a new issue using the bug issue template. Provide all versions numbers and describe exactly what the symptoms are.