bdring/Grbl_Esp32 #153: Connecting PC to ESP32_grbl through Bluetooth

AbySet, 2019-05-26:
Hi everyone, i need help:
I am trying to connect my ESP32 grbl flashed to my PC through bluetooth but i can not do it.
- Using windows 8.1 x64.
- Have ESP32 DEVKIT (Wroom 32 WIFI/BLE).
- Selected in Arduino IDE: ESP32 DEVKIT I.
- I could pair it to my PC and 2 COM ports shows up (COM3 and COM4).
- When i try to use arduino terminal connected to COM3 the arduino IDE freezes right after i send a command like $I.
- When i try to connect arduino IDE to COM4 it does not connect at all and freezes.
- In source code i have disabled the WIFI and enabled the Bluetooth by:
   Uncoment : #define ENABLE_BLUETOOTH
   Comment:  #define ENABLE_WIFI
  I did this in config.h file and then i flash it into ESP32.
- Connecting ESP32 to any Android device and send commands with any serial terminal installed into the device is a succes.

My question is: why it does work on android devices but not on a PC ?
Connecting it to my PC using USB port (wire) is also a succes.

I apreciate any help because i'm planing to upgrade my machine controler from arduino uno to ESP32.

Thank you.

bdring, 2019-05-26:
Check the radio mode with [ESP110] Command from serial port.

See....https://github.com/bdring/Grbl_Esp32/blob/master/doc/Commands.txt

AbySet, 2019-05-27:
Excuse my ignorence but should i flash it with specific file to get to ESP commands ? i have no idea how to.
Any link about that would be nice.
Thank you for your reply.

bdring, 2019-05-27:
I recommend flashing with both Bluetooth and Wifi enabled. You then select whether you use BT or Wifi using the [ESP110] command.

You would send that command via the serial port. I recommend using the Arduino IDE Serial Monitor.

For Bluetooth you would send [ESP110]BT.

If you want live help, you can try me via [Slack at this link](https://join.slack.com/t/buildlog/shared_invite/enQtNjEzNjMwMTA3ODQ2LTJlYjgxYzY1NmE1ZGI2OGQ0NmVmNWU3NzhmNmUwMjBjMTFhNjk1NGE5OWVhMDk0NmIzNTI4NGE5NTE2ZTVlZDE).

AbySet, 2019-05-27:
I got it into ESP commands by activating authentication in config.h
After command [ESP110]pwd=admin   it replies BT.

AbySet, 2019-05-27:
I'll flash it as you said and keep you informed.
Thank you.

AbySet, 2019-05-27:
When i activate BT and Wifi and try to upload it i get this error on arduino IDE:
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board DOIT ESP32 DEVKIT V1.

bdring, 2019-05-27:
You need to change the partition size. See the instructions here and pay particular attention to the partition part...

https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware

AbySet, 2019-05-27:
Ok will do thanks again.

AbySet, 2019-05-27:
I followed the partition size recommandations, activated WIFI and BT in source code (config.h), the upload was a succes.
I could change [ESP110] from AP to BT with succes.
The Bluetooth name of device shown on windows BT devices is "btgrblesp".
The problem on the first post still the same.
Did someone made it working? i just need to know if its possible to do but not working for me.
Thank you.

bdring, 2019-05-27:
Yes, it works. I primarily use it via Android, but here is a screen shot taken today on the latest firmware connected via windows 10.

Try using another terminal than the Arduino IDE. It is sensitive and locks up sometimes.  

[image]

AbySet, 2019-05-27:
WoW thank you so much, this just removed all doubts i had so i can continue investigating until i get it to work.
I will try many terminals and let you know the result.
Thank you.

AbySet, 2019-05-27:
Now i discovered something new:
When i pair ESP32 with the PC it shows "Connected" then after about 5 seconds it shows "Disconnected", i think this is why i can not connect it with any terminal.

AbySet, 2019-05-29:
Ok i was thinking, what about $I value, should it be empty of put something in it ?

bdring, 2019-05-29:
$I is obsolete.  We use [ESP140] now.

AbySet, 2019-05-29:
Thank you for your help @bdring.
Here is my main problem discovered last night after many tests:
- First i'm using Windows 8.1 x64 where Arduino nightly build 1.8.10 is installed, ESP32 driver is installed (Silicon Labs CP210x USB to UART Bridge) and the last version of Grbl_Esp32 is put into Arduino IDE library.
- When i am pairing the ESP32 with the PC it's asking for the BT password, i leave it empty, click "Next" and it does pair it without any problem and i got two COM ports.
- Once paired its shows in windows add bluetooth device page right under the BT device name that is "Connected", **after 5 seconds it becomes "Disconnected"** but the COM ports still there.
- I was using LaserGRBL software to test the connection with the ESP32.
- After pairing the ESP32 with my PC and **within 5 seconds** i quickly connect LaserGRBL then everything is working like it should, if then i disconnect and reconnect LaserGRBL without closing the software, it still working good, but if i close the software and run it again, try to connect its a failure.
- It seems like the ESP32 bluetooth connection, unlike your case is woking just for 5 seconds after getting paired with the PC.
- Still, the ESP32 connects to the phone device all the time without any troubles and serial terminal in it work fine.

At this point i dont know what to think, i was suspecting maybe it should have a password for the BT when used with windows due to windows security stuff and in this case i dont know how to set that into ESP32.

Here is my ESP32 model:
[image]
I added the capacitor to make flashing easier (i had to hold Boot button everytime i flash it, but not anymore with the capacitor).

bdring, 2019-05-29:
Try running the example from the ESP32 Arduino Core.

I don't think this is a Grbl_ESP32 issue.

https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial/examples/SerialToSerialBT

AbySet, 2019-05-29:
Thank you so much your time and your help, i rearly apreciate it, i will investigate more and let you know if i get it to work.

AbySet, 2019-06-03:
Hi again;
I finaly have resolved my issue and its was all about the bluetooth version in the usb bluetooth key pluged into the pc.
- The bluetooth usb key i was using before (the one causing the issue) is version 2.0 .
- The bluetooth version that ESP32 uses is 4.2 .
- The new bluetooth usb key i recently buyed whitch resolved my issue is version 4.0 .

Hope this can help somebody else.

Thank you so much to @bdring for being with me on this issue.

closed 2019-11-05 by bdring