bdring/Grbl_Esp32 #154: Homing for 2 axis question.

teklaser, 2019-05-28:
I have a small 2 axis diode laser that has been working with GRBL1.1 using an Adruino UNO. The homing was setup and has worked well for some time.
I saw the GRBL_ESP32 development come up in a search and I decided to try it out. 
I got the same ESP module and set about making an adapter to hold the ESP and the shield that holds the stepper etc. This upgrade process completed OK.
I have compiled the code from the repository as of the 26th May and uploaded it after setting the correct CPU mapping.  
I have the HOMING CYCLE set to home both the X and the Y axis at the same time as this is what I had on the UNO. 

 #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))

When I try to home the laser machine, the X moves OK, the Y moves OK, the X backs off by the 5mm I have set for the homing pulloff but the Y does not back off.  The system goes into error at that point.  
I checked the code and noticed that axis squaring has been introduced so now the homing cycle runs through some extra code because of the new functionality added.
Is this a bug due to the changes for axis squaring or a feature? 
Should I separate the homing to use 2 cycles rather than the combined setting?

bdring, 2019-05-28:
I'll look into that. It is possible the squaring feature broke that.

Try using separate cycles in the mean time.

bdring, 2019-05-28:
Found the issues.,

By default **#define HOMING_SINGLE_AXIS_COMMANDS** is uncommented in Grbl_ESP32, but not regular Grbl.

That is not allowed with dual axis homing. I did that because I happen to like the single axis homing on my test machine and don't typically dual axis home.

Comment it out to fix that.  

### Do you think I should change it to be more like regular Grbl?

teklaser, 2019-05-29:
I have checked my config.h file and I had commented out that line before I compiled the code that I have used for my testing. 
I will now change to 2 separate homing cycles to see how that goes.
I will let you know how that goes.

teklaser, 2019-05-30:
I have altered the code to use the 2 homing cycles and that has worked OK. 
As for your comment about changing the code to be more like regular Grbl, I personally would like to be able to use the original Grbl homing setting for 2 axis machines but I can work around the issue by separating to 2 cycles.

bdring, 2019-05-30:
I can home X and Y in one cycle on my test machine.  Check out the video.

[Video](http://www.buildlog.net/blog/wp-content/uploads/2019/05/2axishome.mp4)

closed 2019-06-07 by bdring