bdring/Grbl_Esp32 #155: Connecting to diode laser

ggallant571, 2019-05-28:
I have been using ESP32 Grbl on a home brew board for about 2 months with a diode defined by Chinese specs. Bought 15W about a year ago but think it is really 3W It etches wood, slate, & mirrors quite nicely.

Switched to a "quality" 2.5W that has been working for 2 years and the ESP32 went up in smoke. Both are 3-wire from controller to diode (+12V, GND, PWM). Diode behave correctly when jumpered to GND.

2.5W was working on a 5V Arduino Nano. Both diodes will turn full on with a floating control signal and the CPU pulls to GND to turn OFF.

I examined the signals and noticed that the 3W laser is feeding back 3.4V on the control line while the 2.5W module has 5.7V. I have no idea what circuitry is on the modules to supply the floating high or why there is enough current to fry components on the ESP32 board.

There is provision for a 2.2K pull-down on the controller which is intended to keep the laser from firing during boot or reset but it is not installed (lazy me).

I am wondering if:
1. A 3.3V zener diode provide the needed protection.
2. Populating the 2.2K pull-down be sufficient.
3. Be safe and add a level shifter.

easytarget, 2019-10-13:
Option 3. A TTL chip (74 series) is generally recommended. But mosfets and optoisolators are just as good.

Your bigger laser is probably a pure TTL PWM line (high impedance) and cannot overvoltage, nor overload the ESP pin. A 3.3 float voltage is very typical on them. This is the only sort that can be directly connected to 3.3v electronics. And I'd argue that even for this you should have some sort of buffer if using anything more expensive than an ESP chip controller(*) 

The Smaller laser sounds like basic PWM unit, with a resistor pulling the pin to an internal +Ve, 5.7V is a common LED diode voltage. This means it is low impedandce and providing a (relatively) high current at that voltage. More than enough to burn out the ESP's input.

For this unit both your solutions 1 and 2 probably work, but I'd personally not risk it. I'd ass a circuit with an optoisolator to drive this (but I have a bag of optoisolators on my desk, so that's easy for me to say), a simple mosfet or other npn pulldown would work just as well. 

Try to limit capacitance in whatever you do, one of the secrets to clean laser cutting seems (this is just my opinion after seeing it discussed on CNC forums) to be a nice sharp square wave on the Laser. It's the reason why 74 Series TTL logic chips are sometime recommended, they have a very clean output.

ericspooner, 2019-11-16:
Thank you very much for the response. This is where I'm really still learning is when we get to gates and isolators. If I can find something I typicially can copy it, and then after using work to learn to understand it. Would something like this work? [Link to amazon](https://www.amazon.com/CloverUS-Opto-Isolator-Breakout-Optoisolator-Microcontroller/dp/B07XGJVR9Y/ref=lp_306764011_1_14?s=industrial&ie=UTF8&qid=1573909020&sr=1-14) Would I then wire it like the image below? 
[image]

Again, thank you very much for your reply. I will be watching, if you sell the board with SPI communication I will try to be one of the first people in link buying one, as I'm excited to see that.