bdring/Grbl_Esp32 #160: Lowrider z-axis dropping

knchr, 2019-06-05:
I built a lowrider CNC machine and it seems like motion is going to work fine, but I was expecting that the axes would be locked or engaged when I turn the machine on. I can still freely move the head in all directions when the machine is on. It's not so much a problem with X,Y, but with Z, it's not easy to keep or set the home position height. Am I missing something? is it a configuration setting or an option somewhere? I haven't seen much in the way of related posts. If somebody could point me in the right direction I'd appreciate it!

AbySet, 2019-06-05:
When your machine is connected, set $1 to 255 should do it.

Keep in mind that value will keep your motors under power all the time, even when the machine is idle, this mean you will get motors and motors drivers hot all the time (its ok for motors but you may need a fan to cool down the motors drivers in the electronic box).

bdring, 2019-06-06:
IMHO: MPCNC should always use $1=255. The other axes are likely to drift as well.

...hot is OK for motors, but the motors are mounted to PLA printed parts.

knchr, 2019-06-06:
Thank you. That's what I needed. @bdring , I agree MPCNC should always use $1=255. I'll watch the heat for sure.

closed 2019-06-06 by knchr