bdring/Grbl_Esp32 #161: Defining SD pins

just-jason, 2019-06-07:
Is there a possibility to add pin definitions for the the SD card ?
I have fund an ESP32 board in UNO format, and it is a straight plug and play fit on the ESP32. What I am not able to see is how I can define the SD pins to fit my application.

luc-github, 2019-06-07:
it use default values present in variants/pins_arduino.h
```
static const uint8_t SS    = 5;
static const uint8_t MOSI  = 23;
static const uint8_t MISO  = 19;
static const uint8_t SCK   = 18;
```

bdring, 2019-06-07:
@just-jason You don't give us any details on the pins you need to use or the hardware you are talking about.

The ESP32 has several hardware SPI buses. We use the one that Arduino HAL uses. That simplified things for us and makes the code potentially more portable. 

The SD features could be refactored to allow the others, but it is not a priority at this time.

just-jason, 2019-06-07:
@luc-github Thanks, that should be just the thing I need

@bdring , You are right, I have not given you much info to work with, mainly because I am still at the very beginning of an idea, and it is still a bit vague. This is what I can tell you so far.
I would like to use a WEMOS D1 R32 board ( an ESP32 in UNO format ) together with a Protoneer CNC shield which will give me a good plug and play match. I would like to use it together with two stepper drivers to run a x / y plotter, and I would like to adapt the pins to my needs. You have done a great job in the config.h of defining the pins for all the other pins that I need, but I would like to look at moving the SD pins too. 
I will use Luc's suggestion and give it a try

closed 2019-06-07 by bdring