bdring/Grbl_Esp32 #164: Clicking on the spindle button in webui doesn't respond

TurboStudio, 2019-06-11:
Hello, I'm very glad to see this project.In my experiment, I found that the spindle did not respond when I clicked on the spindle button in the web_UI. I don't know what's wrong there. When I send the M3/M5 command, the spindle can move normally.

luc-github, 2019-06-11:
Spindle button send :  0X9E per @bdring specifications , he says 
>that toggles the spindle on and off in a hold state
>If you are running a CNC and you do a feed hold, the spindle stays on. If you want to toggle it on/off, you use that button

>If you turn it off, it will turn itself back on if you start the job and forget to toggle back on.

bdring, 2019-06-11:
That part of the WebUI is the overrides.

There is no spindle toggle in Grbl. You could add dedicated spindle on and spindle off macro buttons to get what you want.

TurboStudio, 2019-06-11:
> WebUI的那部分是覆盖。
> 
> Grbl中没有主轴切换。您可以添加专用主轴和主轴关闭宏按钮以获得您想要的。
OK,Thank you very much！

TurboStudio, 2019-06-11:
> 他说，主轴按钮发送：每个@bdring规格为0X9E
> 
> > 在保持状态下打开和关闭主轴
> > 如果您正在运行CNC并进行进给保持，则主轴保持打开状态。如果要打开/关闭它，可以使用该按钮
> 
> > 如果您将其关闭，如果您开始工作并忘记切换回来，它将重新打开。

Thank you very much！

closed 2019-06-14 by TurboStudio