bdring/Grbl_Esp32 #166: [roadmap] IO expander and gpio Idea

Tonymac32, 2019-06-14:
Didn't have another means of contact, I'd like to make a Hat for my upcoming Pi "A" shaped ESP32 board to support this, and it happens to use an I2C GPIO expander to fill out the 40-pin header.  I won't link to it here unless you're ok with it, it's soon-to-be-launched crowd sourcing.  (no matter the outcome of the crowd sourcing it is being released under CERN OHL as open source hardware)

I'll have to take a closer look at the schematics.

And who knows, maybe some single board computer guys would like a Hat with that stuff on it.

bdring, 2019-06-14:
Is this project designed to, or able to run Grbl_ESP32? If so, I do not mind you posting a link. 

BTW: We like to close issues, so after a while, if no discussion is going on, the issue will get closed.

Tonymac32, 2019-06-14:
It's a 16 MB ESP32-WROVER-B based board in a so it would run the software,
now that I know this project exists I will design a HAT for it to go with
my board. (Open source hardware)  It's not out of the box yet though so
I'll not further spam your github until I at least have a prototype
available of the hat.  Got excited to see an upgrade path for my CNC.  :-)

As to the issue closing, certainly.