bdring/Grbl_Esp32 #170: Losing connection due to interference.

mac7988, 2019-06-19:
Hi,

I am losing Bluetooth connection to my GRBL sender when some other machine turns on in the same space. 

I noticed while an axis is in motion if a saw near me turns on it stops the motion. I would have to reset it and Rehome. Is there something i can do to fix this noise problem?

thank you in advance

bdring, 2019-06-19:
Can you hook it up to USB and post the data you get via serial when it stops?

mac7988, 2019-06-19:
I can't as it is using the usb to power the card. I can provide the verbose ouput.

AbySet, 2019-06-19:
What are you using to connect the machine to ? is it Android device or PC, if PC is it laptop or desktop ?
Some usb bluetooth 4 keys have ability to be connected with more than one device and they are a perfect solution to this kind of troubles.

mac7988, 2019-06-19:
I am using an Android tablet.

AbySet, 2019-06-20:
- Check the tablet specs to see if it is using bluetooth 4 or any other (bluetooth 2 maybe).
- Some bluetooth 4 devices (but not all of them) are able to handle more than one device at same time but not bluetooth 2 devices.
- If your tablet is not capable of that, you should use one machine at the time, that mean the unused machine should not just be stoped but also disconnect from tablet bluetooth.
- In some rare cases, wifi can also disturb bluetooth connexions, some BT devices have a feature called "WLAN coexistance interface" to prevent that from hapening.

bdring, 2019-06-20:
Are you sure the ESP32 is not resetting and it just looks like loss of bluetooth?

mac7988, 2019-06-20:
Hi all,

The Esp is resting.

bdring, 2019-06-20:
I doubt it has anything to do with bluetooth.

Sent a photo of you setup.

gerritv, 2019-07-28:
If the ESP end is resetting, then itis the power source you need to sort out. Add a filter capacitor to 5V supply. Or run the controller from an extension cord to a different wall plug, perhaps from another room. Worst case run the controller off a UPS, it might filter out the spikes. Electrical spikes from motor start ups will travel through many inexpensive supplies.

Gerrit

closed 2019-08-23 by mac7988

AbySet, 2019-08-23:
mac7988, if you have resolved the issue please let us know how, it can help others.
Thank you.