bdring/Grbl_Esp32 #171: New PCB to Test Features

bdring, 2019-06-21:
I am working on a test PCB for some upcoming features.

[image]

Features working on...

1. SPI drivers in daisy chain mode. This means only one CS is need for all drivers.
2. Diag pins are routed back to ESP32 for stall detection.
3. Optionally use SPI stepper enable rather than wired pins.
4. (4) independant axes. I'll probably update the code for up to (6)
5. Larger caps (330uF 50V) for high current TMC5160 drivers.

This is just a test bed for trying out new features that may or may not make it into the master branch.

AbySet, 2019-06-23:
Great job, it is just what i was looking for.
Is it open source? if so, where can we find schematic, pcb and esp32 firmware for it ?
I would like to put this in my cnc machine as soon i get it.
Thank you.

bdring, 2019-06-23:
I will be ordering about 10 raw boards and a solder stencil later today. The controller will not work until the firmware changes are done.

The most important change is the SPI cascade mode.  This requires the [TMCStepper Library](https://github.com/teemuatlut/TMCStepper) to add that feature. There is [an issue](https://github.com/teemuatlut/TMCStepper/issues/42)  open on that. There is no timeline for that, so I may need to temporarily just send raw bytes to the SPI rather than the much more friendly and readable library.

The extra axes need to be added to Grbl_ESP32. I have done that on other Grbl ports, so it should not be much of a challenge. I will have an option where two of the axes can be ganged/squared so ShapeOko/X-Carve style machines can benefit.

AbySet, 2019-06-23:
Thank you so much, i will be waiting for them (firmware and GitHub reposity).

bdring, 2019-07-03:
[image]

I got some basic daisy chained SPI drivers features working. I only have two of the TMC5160 drivers. I just hacked the TMCStepper Library a little to make it work. Right now it just sends the same config to all drivers.

AbySet, 2019-07-03:
WoW, i'm so exited right now, my machine still waiting for that.
Please let me know when all is ready to go, i think i will engrave the pcb with cnc or probably just use the old method (chimical).
I have no words to thank you for all the efforts.
Once my machine is ready i will daily used it and it will be a permanent test for this project.
Thank you so much.

MtripleM, 2019-07-05:
> I am working on a test PCB for some upcoming features.
>
> [image]
>
> Features working on...
>
> 1. SPI drivers in daisy chain mode. This means only one CS is need for all drivers.
> 2. Diag pins are routed [...]

How many maximum axis for ESP32 ?

bdring, 2019-07-05:
I have had Grbl running up to 6 on a different port. 

The current Grbl_ESP32 is limited to 3, but I will add 3 more soon.

shawe, 2019-07-14:
Only to my knowledge, at this time this PCB+ESP32 is similar to this other alternative https://reprap.org/wiki/MiniRambo ?

And in other hand, the increase in CPU power ESP32 vs Arduino, it implies in a fastest work?

misan, 2019-07-15:
While both boards can control several stepper motors with software to
decode g-code the two boards are not related anyway. Not only they have
different processors (8-bit miniRambo vs 32-bit ESP32 ) and vast
differences in RAM and Flash, but miniRambo is aimed at 3D printers and
PCB+ESP32 is aimed at CNC machines up to 4-axis and/or laser
engraver/cutter.

Or course, a faster processor means the pulse max frequency can be much
higher.

bdring, 2019-07-15:
The ESP32 was selected for this project primarily for the Bluetooth and WiFi features, but it has proven to be excellent in other areas.

- Very Fast. This can generate high step rates. Steppers can only go so fast before they are useless, but servo motors can use the higher rates. The speed and dual cores also allows background operations, like web server, kinematics, etc to operate without affect on the motion quality.
- RAM The ATMega chips are severally limited. 
- Peripherals . The RMT, high speed/resolution PWMs, timers, DACs, UARTs, etc are great and have super flexible pin mapping.
-Cheap and Modular. The small and cheap dev modules are great.

shawe, 2019-07-19:
@misan @bdring thank for your replies.

I have an ESP32 also, but from Espressif and pin connections are a little different (some extra pins), but it isn't really a problem.

JackHouseLab, 2019-08-18:
Hi!
I hope it is right place for my post :-)  (don't see email or private message for autor of Grbl_ESP32).

I have designed another version of PCB for GRBL ESP32 project - this time it uses optocouplers on inputs, and pnp transistors on outputs. It is intended to use with external stepper motor drivers like DM5042 or simillar which have optocouplers on its (STEP, DIR ENABLE) inputs. 
PCB will be placed in Gainta D6MG enclosure with screw terminal connectors to easy connect to external devices.
This project use ESP32_DEVKITC_32D version of Espressif module.
Here is some views of this project:
[image]
[image]
[image]
Project is made in Kicad and Freecad, and is of course intended to be available for people as Open Hardware Project.  
It uses current version of software and have functionality limited to this version (3 axes only - there is no more pins on ESP32 to make additional axes without adding more IC's - like I2C expanders or something like that).
First PCB's will be ordered (probably) tomorrow - just a few for testing.  
I can send project files to author - to make them available for people (for adding to project as alternative or  something) if he wants and is planning such cooperation - of course :-)
This PCB project needs some testing in real word of course - for now it is just virtual project. 

Best Regards

misan, 2019-08-18:
@JackHouseLab I guess you already answered this when you mentioned no more pins left, but a chance to keep an SD card in the board, even if pins were shared with a second function might come in handy.

At any rate, a very nice addition for the larger CNC machines.

JackHouseLab, 2019-08-18:
A propos SD card....It is hiding under ESP32 module (still exist) :-)
[image]
Some of pins are less useful - like "door", flood", "mist" - they may (probably) be reprogrammed and used as 4th-axe. 
Don't know if  output signals on STEP pins are realised only by software or use some internal hadrware, and any pin may be used to this task.  There is about 100kHz on step pins.

misan, 2019-08-19:
> A propos SD card....It is hiding under ESP32 module (still exist) :-)

@JackHouseLab That is great then.

stanw, 2019-08-20:
It looks like you have driver transistors on the step and dir lines. Do they pull up or down? If they pull down, a nearby +5V port will be handy to keep the wiring and loop area minimized to reduce noise and interference.

Looks good.

Stan

JackHouseLab, 2019-08-20:
Yes - driver transistors pulls down.  You can take +5V from 7805 located in the PCB (or external power supply) and use it to drive optocouplers or relays, etc...

Here is schematic of this board:
[Grbl_ESP32.pdf](https://github.com/bdring/Grbl_Esp32/files/3521992/Grbl_ESP32.pdf)

PWM transistors can be powered from external power supply too - if there is a such need.
BC817 can drive up to 500mA and up to 50V (Vce).

stanw, 2019-08-21:
Looks good. Where do I sign up to get one :-)?

Stan

JackHouseLab, 2019-08-21:
Hi,
I ordered a few PCBs from my local PCB manufacturer, as a part of bigger order - it will be ready for about 3 weeks.  Then I will assembly one, and check if everything works correctly. 
If yes, I will publish gerber files to community. 
I'm not planning to sell boards - you may order PCBs from your local manufacturer, or maybe someone near you do it, and sell to other people...and maybe there is another way....

I prefer developing than selling ;-)

xP373Rx, 2019-09-24:
I use 2 74HCT245 to drive the optos on my stepper  drivers.
http://gsgrid.net/gcode/grblesp32.jpg
http://gsgrid.net/harb/harb/PitsIphone/Alle%20Fotos/IMG_5453.JPG
http://gsgrid.net/harb/harb/PitsIphone/Alle%20Fotos/IMG_5454.JPG
Works slike a charm. 
Also hacked the Nextion display to control the esp.
http://gsgrid.net/harb/harb/PitsIphone/Alle%20Fotos/IMG_5457.JPG
http://gsgrid.net/harb/harb/PitsIphone/Alle%20Fotos/IMG_5456.JPG
http://gsgrid.net/harb/harb/PitsIphone/Alle%20Fotos/IMG_5455.JPG

JackHouseLab, 2019-09-24:
Looks great :-) 
Is display connected with ESP32 via cable or wireless maybe? 

My board in real:
[image]
Not conneced to machine yet, but I did some tests by oscilloscope, and everything seems good. 

As I can see, ESP32 module takes more current, when it is in AP mode (140mA) than Station mode (about 70-80mA). 
So I added small heatsink to 7805 IC (it looks bigger on the picture than in real ;-) )
I had to uncomment some lines in config files to get Hold/Resume/Abort lines working.

Now it's time to replace old arduino grbl controller by new one, and check how it works :-)

JamesNewton, 2019-09-24:
VERY nice looking board there Jack. Pretty much "shut up and take my money" nice. Only thing I would add is PMinMO headers for open source drivers. 
http://www.massmind.org/Techref/io/PMinMO.htm

JackHouseLab, 2019-09-24:
Thanks :-)  

I connected to machine , but have problem with homing (settings are the same as original grbl has) - after about 2mm of move I have such message: 
 Alarm
Homing fail. Pull off travel failed to clear limit switch. Try increasing pull-off setting or check wiring.

After increasing pull-off setting to 200 (from 2), the move was continuous (for X-homing) and the limit switch didn't worked, and only hand reset solved me from damage of machine. 

Is something missing (commented out) in the compilation?

bdring, 2019-09-25:
It is good to check the limit switches manually by looking at the ? response.

misan, 2019-09-25:
LEDs on inputs and outputs may come in handy for troubleshooting.

JackHouseLab, 2019-09-25:
Now everything seems to work perfectly :-)
I just have checked probe, and it works OK too.
Now I have to change wires and remove some of not necessary connectors, to make sure of good connections beetwen controller and rest of electronic modules, etc...

I made some changes in config files, but each time I have to remove ESP32 from socket, to flash it properly.  When it is inserted in board, I cannot reprogram it (message from Arduino IDE about bad checksum after programming).  
It is common or only my problem? 
Fortunately enclosure have removable top lid, so I have quite good access to ESP32 ;-)

Attaching LEDs to inputs and outputs is a good idea - it would be up to 20 LEDs (with Power LEDs).
In next version of PCB I could make some changes, add connectors and even design front panel PCB with SMD LEDs for easy soldering.

AbySet, 2019-09-26:
**JackHouseLab** said:
 "When it is inserted in board, I cannot reprogram it (message from Arduino IDE about bad checksum after programming)."

This happends when RX and TX are connected to the main board and are not (NC) whitch disturbs usb data, but i'm not sure whitch gpio they are connected to, see here:
https://github.com/eerimoq/simba/issues/137

misan, 2019-09-26:
> eerimoq/simba#137 <https://github.com/eerimoq/simba/issues/137>

Certain pins have to be low or high when uploading code according to the
information in here:
https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

JackHouseLab, 2019-09-30:
Thanks for the links :-)
I will check which pin is making bad work, but I have to make another controller board for this, because this one is sitting in my machine:
[image]
It works 3 times faster than original arduino based grbl CNC controller :-)
 
It (probably) would be better to make Probe input as internally connected to Vdd, and switched on by shorting input (cathode of internal LED diode in optocoupler) to GND, than anode to Vdd, as is done now. 
It would allow you to GND your spindle (now I can't do this, but it works with +12V anyway).
I'm really not sure which way is better :-)

closed 2019-11-05 by bdring

jasinghuang, 2019-11-07:
> Thanks for the links :-)
> I will check which pin is making bad work, but I have to make another controller board for this, because this one is sitting in my machine:
> [image]
> It works 3 times faster [...]

Hi JackHouseLab, nice design! Would you share the design file or pcb doc? I also want to make one. Thanks

misan, 2019-11-07:
> I made some changes in config files, but each time I have to remove ESP32 from socket, to flash it properly. When it is inserted in board, I cannot reprogram it (message from Arduino IDE about bad checksum after programming).
> It is common or only my problem?
Some GPIO has to be floating (or biased) for the ESP32 to be programmed. If the input or output bias blocks this then programming cannot take place. (It happens the same with CNC shield enable resistor R1). 
Better explained here: https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

xP373Rx, 2019-11-07:
Pin 12 on esp 32 has to be floating while boot up

JackHouseLab, 2019-11-09:
> Hi JackHouseLab, nice design! Would you share the design file or pcb doc? I also want to make one. Thanks

Hi,
Here are Gerber and Kicad files:
[GRBL_ESP32_Kicad_Files.zip](https://github.com/bdring/Grbl_Esp32/files/3827604/GRBL_ESP32_Kicad_Files.zip)
[GRBL_ESP32_gerber_files.zip](https://github.com/bdring/Grbl_Esp32/files/3827605/GRBL_ESP32_gerber_files.zip)

vitalibr, 2020-05-30:
@JackHouseLab you PCB is amazing!

I would like to use it with my stepper motor drivers, but with 24v.
I saw that you are using BC817, I could use them with 24v without a problem, right? 

Should I put some resistor between external power supply 24v and collector?

Thanks

vitalibr, 2020-05-31:
Perfect pcb @JackHouseLab !

I would like to use your project, but I have some stepper motor drivers that the manufacturer recommends using 24v for dir/stp/en signals. I would like to keep that and use 24v.

How can I do that? 
[image]
[image]

does one of the two make sense? Sorry, I understand the basics of electronics

the vcc being powered by a 24v source

JackHouseLab, 2020-05-31:
Hi :-)
BC817 can drive up to 50V (Vce), so, you can use it for 24volts.
In my project,  cathode of internal optocoupler LED  (which is in stepper motor driver) is connected do BC817 collector, and anode of LED is connected to '+' of power supply.
But it depends from type of your stepper motor driver - if it have optocouplers on inputs or not, and how they are internelly connected.

vitalibr, 2020-05-31:
> BC817 can drive up to 50V (Vce), so, you can use it for 24volts.
> In my project, cathode of internal optocoupler LED (which is in stepper motor driver) is connected do BC817 collector, and anode of [...]

interesting , unfortunately the manufacturer  does not say these details, only that I can supply 3.3v-24v (Stp, dir and en input), I believe that it uses optocoupler just like your driver, maybe I need to remove the aluminum case to confirm taking a look on the driver pcb.

what is your driver model @JackHouseLab ? (curiosity)

MitchBradley, 2020-05-31:
The basic problem here is that you are confusing the motor voltage and the optocoupler voltage.  They are completely separate.  The whole purpose of having optocoupler inputs on motor driver modules is so you can use a high voltage to drive the motors, while controlling from a lower voltage, lower current microprocessor system.

Okay, let me explain how an optocoupler works.  An optocoupler is an LED, usually infrared, and a phototransistor in the same package.  When you energize the LED, it lights up and shines on the phototransistor which starts to conduct.

To light up an LED, you have to push current through it.  The right amount of current is not precisely defined because the tolerances of the components inside an optocoupler is pretty wide.  But it is generally the case that 10 mA through the LED will always be enough to turn on the phototransistor, and very often you can get by with as little as 2 or 3 mA.  5 mA works in 99+% of cases.

Okay, so how do you control the current through the LED, given that you are usually switching voltage to it?  You have to put in a series resistor.  The LED itself has a fixed voltage drop of roughly 1.4 V (the exact drop is the Vf parameter in the optocoupler data sheet and it varies a little bit with current, but it does no good to worry about the exact value since the overall sensitivity of the optical system is so poorly controlled).

The current through the LED is (driving voltage - Vf) / series resistance.  If driving voltage is 5V and Vf is the usual 1.4V, then to get 10 mA you need   (5V - 1.4V)/R = 0.01  giving R = 360 ohms.

Most, if not all, optocoupled drivers have a built-in series resistor in the range of 270 ohms to 330 ohms.  With that value of built in series resistance, you can usually get by with driving them either from 3.3V or 5V.  At 3.3V, the LED current will be a little low but the system will still work.  At 5V the LED current will be a little high but it will still work.

If you try to drive with a higher voltage, say 24V, unless you add extra external series resistance, you will blow up the LED.  The current would be (24-1.4)/270 = 83 mA which would make the LED overheat.

So if you want to use 24V volts, you would need to increase the resistance to (24 - 1.4)/R = 0.01 giving 2260 ohms, so you would need to add an extra 2K resistor in series with the one that is already there.

There is no reason whatsoever to use 24V to drive the optocouplers!  Just use 3.3V or 5V with the optocouplers.  It is good to drive the *motors* with 24V, but the motor voltage is different from the optocoupler voltage.

vitalibr, 2020-05-31:
@MitchBradley thank you so much!! an excellent explanation

I know the difference between motor voltage and the optocoupler voltage. Now I’m using 48v to motor (like the manufacture recommended)
I would like to use 24v on optocoupler voltage b because I read that industrial devices use 24v to avoid problems with noise. I have a 3kw vfd and spindle with emi filter but I am afraid that I still have noise issues, wouldn't 24v be safer?  why do big machines only use 24v?

vitalibr, 2020-05-31:
what the manufacturer says in the manual
“ The pulse level that R60 drive can be used: 3.3V-24V (no need to connect resistor)”

[image]

https://drive.google.com/file/d/1ub8DBp7_Yz3SZPcCXhtoHxsbA14Td3Sk/view

MitchBradley, 2020-05-31:
Start with the easy way and see if you have problems.  If you have problems you can always change it.

If the manufacturer says you can use between 5 and 24V with no additional external resistor, then it will probably work.  Maybe they are using high-sensitivity optocouplers.

Changing the BC817 driving circuit is not going to change anything.  It is already rated at more than 24V.  The only variables are what voltage you use on the + side of the optocoupler, and whether you add additional series resistance.  If it were me, I would start with 5V and no extra resistance, then spend my time doing very clean wiring, with twisted pairs going to each control input, keeping the control pairs away from the spindle wiring.

MitchBradley, 2020-05-31:
Based on the photos of Rtelligent R60 on aliexpress, the control inputs have both + and - terminals.  That means they are optocoupled.  If they were voltage driven, not optocoupled, there would be a shared ground terminal and a single input for each signal.

You are overthinking this and going off into the weeds.  This rtellignet driver looks to be substantially the same as other external drivers that are routinely used.  The way people use such drivers, which works fine, is either to connect 5V to the + terminal and switch the - side to GND with a transistor, or to connect the - side directly to ground and switch the + side to 5V with a high-side driver circuit.  The first way is appropriate for the jackhouselab board.

JackHouseLab, 2020-05-31:
"what is your driver model @JackHouseLab ? (curiosity)"
I'm using D5042 stepper motor controller.
Here is recommended schematic of connections between stepper controller and Grbl ESP32 controller:
[image]

vitalibr, 2020-05-31:
thanks for the support and patience @JackHouseLab and @MitchBradley , 

Last (noob) question, 
with the BC817 it will not reverse the low and high logic as it is an NPN?  Shouldn't it be PNP?

or is this being done in the code?

mutumbik, 2020-09-28:
Another option under the external drivers

[image]
[image]
[image]
[image]

JackHouseLab, 2020-09-29:
Good work :-)  Looks simmilar to mine ;-)

I completed three boards until this time.  All works fine.  
The last one is connected to LPT port of CNC machine (the older one, which don't have USB port), and thanks to this machine works via USB port or WiFi - with help of GRBL sender (i.e. Candle). 
The only thing I had to do was remove BC817 transistors, and connect ESP32 ports directly to LPT inputs.  Level of 3,3V from ESP32 pins is sufficient to drive 74HC245 inputs without any problems. 
So, this way you don't have to search and buy computer with LPT port to drive such CNC machines.  No internal soldering or devastating PCB is needed.

Althafzahra, 2021-09-17:
> Hi,
> Here are Gerber and Kicad files:
> [GRBL_ESP32_Kicad_Files.zip](https://github.com/bdring/Grbl_Esp32/files/3827604/GRBL_ESP32_Kicad_Files.zip) [...]

do you have facebook?  I want to ask more about esp32 grbl

KeDP72, 2022-08-05:
> Hi, Here are Gerber and Kicad files: [GRBL_ESP32_Kicad_Files.zip](https://github.com/bdring/Grbl_Esp32/files/3827604/GRBL_ESP32_Kicad_Files.zip) [...]

Hello!
I wanted to ask, maybe you have finalized your design and made some kind of improvement? How does the board behave during all this time?
I want to repeat your development, with a couple of additional axes