bdring/Grbl_Esp32 #173: Best Way to Keep Current with ESP32-GRBL Developments?

LHC-Canada, 2019-06-22:
Hello Bart - 

I recently became aware of your ESP32-GRBL developments and am very interested in following along with your progress on new features etc.  At the moment it seems there is an "issue" thing here that I can get to via your tindie listing for the 3 channel PCB, but there's a MPCNC PCB listing that has no links to here.  A google yields some buildlog blog thing, and also a twitter thing that seems to have postings from you that are very recent. Are there any other places to "watch from the bleachers" or are these the best ones ?  

I'm retired and I've been considering building a CNC router (possibly MPCNC) this winter and your ESP stuff looks fantastic for the features and capability.  I also work with a local university makerspace that has an MPCNC they are currently trying to bring back to life as it was in storage and then donated to them.  They are interested in your controller designs as well.  There might also be a possibility I might be able to find some talented students that could help support this great firmware development as to to features etc so I am going to ask around.

Anyway, thanks very much for the wonderful contribution to the open source CNC controllers!

Cheers from Canada,
Lewis

bdring, 2019-06-22:
The github repo for the MPCNC is [here](https://github.com/bdring/Grbl_ESP32_MPCNC_Controller).

There is also a [Slack for Grbl_ESP32](https://join.slack.com/t/buildlog/shared_invite/enQtNjEzNjMwMTA3ODQ2LTZkMTE5MjcwZTY4MjgwOWEwMWE1MzZlNGNhNDdmMWEwNWZkN2Y5NjBlNjBlZjY2MTg3ZThlN2I3YjEwMzUyMWI).

gerritv, 2019-07-28:
Lewis, you can also Follow all of @bdring 's projects by clicking on his name at top of this page and then selecting Follow under his photo.
Gerrit

closed 2019-09-10 by bdring