bdring/Grbl_Esp32 #179: Can't get GRBL to fit on ESP WROOM 32?

disneytoy, 2019-07-03:
In the current Arduino IDE, I'm trying to install on a ESP WROOM 32. But I get the error that it is to large?

Arduino: 1.8.8 (Windows 7), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 115200, None"

Sketch uses 1775330 bytes (135%) of program storage space. Maximum is 1310720 bytes.
Global variables use 68724 bytes (20%) of dynamic memory, leaving 258956 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board DOIT ESP32 DEVKIT V1.

bdring, 2019-07-03:
Read the [wiki on compiling the firmware](
https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware) regarding partitians.

disneytoy, 2019-07-03:
After some research, in Arduino I changed PARTITION TYPE: to HUGE APP (3mb no OTA) and it seems to compile! Just wasn't familiar with the partition types.

Build options changed, rebuilding all
Sketch uses 1775314 bytes (56%) of program storage space. Maximum is 3145728 bytes.
Global variables use 68724 bytes (20%) of dynamic memory, leaving 258956 bytes for local variables. Maximum is 327680 bytes.

closed 2019-07-03 by disneytoy

bdring, 2019-07-03:
It is better to use the partition we recommend. It gives you OTA and SPIFFS for WebUI