bdring/Grbl_Esp32 #180: Wall V-Plotter

phantomse, 2019-07-03:
In my search for an hanging all in one Wall V-Plotter. Always skipt this project thinking it was only for round coasters. I wanted to make it wireless (Battery or Power-line only, or get its power from the two cables it hangs from), much easier to set up and calibrate no matter of the size of the surface.

Hardware: 
- the hardware I wanted to keep simple since V-Plotter doesn't need much
- haven't tested it yet so I hope to get some advice
D1 R32 (ESP32 UNO) (only one pin seams to be incompatible, snippet it off) + CNC Shield V3.0
Stepper + Servo + Solenoid + Endstops + SDCard ... => auto color changing and advanced functions
all in one, similar to https://hackster.imgix.net/uploads/attachments/536161/img_0131_joZbk8YliZ.JPG?auto=compress%2Cformat&w=1280&h=960&fit=max

Luc is maintaining a version of Marlin 2.0 with ESP3D integrated but only Marlin 2.1 will get polar kinematic ( still years ... )
https://github.com/MarlinFirmware/Marlin/issues/5275
there are also old versions of Repetier and GRBL with polar kinematic but non esp32 compatible.
https://github.com/jasonwebb/grbl-mega-wall-plotter
https://github.com/RickMcConney/PenPlotter
Polargraph is trying to do all from scratch on ESP32 and thereby has little progress.
We have the hardware only the software lags behind!???

Tried compiling it in atom PlatformIO after adding some needed Library's 
platformio.ini
board = D1 R32 ??? (not yet configured)

davidelang, 2019-07-30:
The Maslow CNC project is looking to do a next-gen controller board, and one of the options we are looking at is to base it off of this project. It would require adding support for the v-plotter style kinematics, and to add support for DC motors with encoders instead of steppers.

Is this sort of thing something that you would welcome as options? or something you think would need to be maintained as an independent port?

bdring, 2019-07-31:
I am definitely interested in options like that. It might be easiest for you to create a fork. That gives you complete control over it until it is ready to be pulled back into the master branch.

What type of DC motors are you looking at?

madgrizzle, 2019-08-01:
I think these are the stock motors being used today:

https://www.alibaba.com/product-detail/ET-WGM58AE-with-7PPR-encoder-12v_60665360361.html?spm=a2700.8304367.0.0.PlKjgw

The current controller uses either L298 (original design) or TLE5206 (newer design) but there's further contemplation of using a different h-bridge for the new(est) controller.

i-make-robots, 2019-08-18:
Hey all.  I make the http://makelangelo.com/  polargraph.  The firmware is Marlin-based.  We have some tentative support, but (I'm told) it's throwing a watch dog timer error at run time.  If any of you would be interested, the project is open source and well established.  The firmware is available at https://github.com/MarginallyClever/Makelangelo-firmware/.  Cheers!

ithinkido, 2019-08-18:
https://github.com/luc-github/Marlin/issues/9#issuecomment-508148547

rockets-cn, 2021-07-01:
I found https://github.com/ilaro-org/grbl-polar
the question is how can I merge the code into ESP32 one.
It's seem that we have to build a custom cpp and Machines for that.

rockets-cn, 2021-07-12:
https://github.com/rockets-cn/esp32-wall_V-CNC
my friend and I try to make it. 
already build a custom cpp and Machines.
Welcome to check.

rockets-cn, 2021-07-22:
I have tested the code. it's work!

CCERocks, 2021-11-22:
> I have tested the code. it's work!

do you have video of working machine ?

rockets-cn, 2021-11-23:
not yet, we have tested without the cut. but not uploaded yet.
Thank you test for us.

i-make-robots, 2021-11-23:
Makelangelo v plotter now runs Marlin.  Updated link in previous comment.
Anyone have the STL file for Bart's 3-jaw-chuck pen holder?  that looks really interesting.

rockets-cn, 2021-11-26:
OK，I will go to check.