bdring/Grbl_Esp32 #185: not compiled

koka7777745, 2019-07-25:
does not compile, gives an error
[image]

luc-github, 2019-07-25:
it compile fine with standard config : https://travis-ci.com/odaki/Grbl_Esp32/builds
function exists : https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Esp.cpp#L168

what change did you do in FW / config ?
what version of esp32 core are you using ?

koka7777745, 2019-07-25:
I did not change anything in the configuration, I wanted to compile the firmware to upgrade the old version to ESP 32

luc-github, 2019-07-25:
what version of esp32 core are you using ?
what version of grbl_esp32 are you compiling ?

koka7777745, 2019-07-25:
I compile from the last downloaded source, for an esp-wroom 32 board

luc-github, 2019-07-25:
well with such info I only can say : 
* delete all 
* install again latest git of esp32 core , the development repository : https://github.com/espressif/arduino-esp32#installation-instructions
* download again latest git of grbl_esp32 master
* copy libraries as explained in wiki
* compile with fresh install

koka7777745, 2019-07-26:
All deleted and cleaned, and installed on a new one. It still gives errors. Can you tell me?

Arduino: 1.8.9 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

sketch\grbl_bluetooth.cpp.o:(.bss.SerialBT+0x0): multiple definition of `SerialBT'

sketch\BTconfig.cpp.o:(.bss.SerialBT+0x0): first defined here

sketch\grbl_eeprom.cpp.o: In function `memcpy_to_eeprom_with_checksum(unsigned int, char*, unsigned int)':

sketch/grbl_eeprom.cpp:23: multiple definition of `memcpy_to_eeprom_with_checksum(unsigned int, char*, unsigned int)'

sketch\eeprom.cpp.o:sketch/eeprom.cpp:23: first defined here

sketch\grbl_eeprom.cpp.o: In function `memcpy_from_eeprom_with_checksum(char*, unsigned int, unsigned int)':

grbl_eeprom.cpp:(.text._Z32memcpy_from_eeprom_with_checksumPcjj+0x0): multiple definition of `memcpy_from_eeprom_with_checksum(char*, unsigned int, unsigned int)'

sketch\eeprom.cpp.o:eeprom.cpp:(.text._Z32memcpy_from_eeprom_with_checksumPcjj+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\KOKA\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
Multiple libraries were found for "SD.h"
 Used: C:\Users\KOKA\Documents\Arduino\hardware\espressif\esp32\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

luc-github, 2019-07-26:
yes - you did not delete but copied new grbl_eps32 on old grbl
there is not more `eeprom.cpp` in FW but now it is`grbl_eeprom.cpp`

just do a real clean install - delete all previous code before installation

koka7777745, 2019-07-26:
Thanks, it turned out to compile, it works.

luc-github, 2019-07-26:
Great. 
Please do not forget to close issue.

closed 2019-07-26 by koka7777745