bdring/Grbl_Esp32 #187: Allow jogging with door open

Geekoid85, 2019-08-11:
Okay so I finally take the time with some friends of mine to make the esp32 into a 1997 cnc router. I found some quit annoying behaviour that can maybe concern other users.

Because this cnc machine is in a public workspace (fablab in France) safety is a priority. So I would like to use the old door switch. Currently, it's not wired up because we found it very practical to setup xyz origin with the door open. So you guess it, grbl does not allow jogging with door open. By default it's great, but an option in the config.h to ignore the input pin during jogging can be a good feature right ?

One thing is sure, grbl esp32 is a big upgrade for this machine!

bdring, 2019-08-11:
I can look into it. This is the type of thing I don't like to deviate from regular Grbl on though.

Geekoid85, 2019-08-11:
Yeah that's a bit tricky I understand. Thinking about it a $$ setting will be ideal. So for exceptional case it is possible to completely disable the door switch and run the machine open. It's better than a physical switch, less accessible for not initiated users.

bdring, 2019-08-11:
I think it is best to stay with the current Grbl rules. 

Can you solve the problem with wiring?

- A key switch that overrides the door circuit?
- The door only interlocks the spindle?

Geekoid85, 2019-08-12:
Yep I can solve the problem in a hardware way. I can add a relay in parallel to the one use for spindle enable so that the door switch only work when spindle enable is turn on, so when a job is running. To override this feature (in case we need to work with the door open) I can add another switch in series with the relay.

So yeah It's doable, just asking why "grbl" has potentially made the choice of not allowing the user to configure the door behavior so that it is possible to jog the machine with the door open.
Maybe It's a good idea to open an issue on the grbl github. So that this project stay in line with the original grbl project.

closed 2019-09-10 by bdring