bdring/Grbl_Esp32 #188: 4 axis high current CNC controller

aldanstar, 2019-08-12:
I saw that 4 axis high current CNC controller is builded by you about month ago. When PCB will be officially released with updated firmware? Is firmware will be uploaded to this repo or new?

bdring, 2019-08-12:
I will release the source files soon. I am on vacation for another week. 

The firmware will be incorporated into master branch eventually. It is on [this branch](https://github.com/bdring/Grbl_Esp32/tree/TMCStepper) now.

It uses SPI drivers in daisy chain mode. You must use [my fork of the TMCStepper library](https://github.com/bdring/TMCStepper).

bdring, 2019-08-12:
@aldanstar 

I created [a repo](https://github.com/bdring/4_Axis_SPI_CNC) with what I have so far.

closed 2019-08-19 by bdring