bdring/Grbl_Esp32 #194: Dir and Pulse on Grbl_ESP32 CNC Development Board V3.5, Are They 5V or 3V?

jaysettle, 2019-08-21:
Wanting to use DM542T stepper drivers with the ESP32 running GRBL. 

1. I don't mind using a PCB if it gets me 5V level shifters for axes and spindle and caps for limit switches etc. **Is the Grbl_ESP32 CNC Development Board V3.5 the board I should go with?**

2. I've found that the spindle pins are level shifted. **Are the Direction and Pulse pins level shifted to 5V on the Grbl_ESP32 CNC Development Board V3.5?**

3. Looking at the picture below, is this the correct way to wire this from the Grbl_ESP32 CNC Development Board V3.5?
[image]

[image]

https://www.amazon.com/gp/product/B06Y5VPSFN/ref=ppx_yo_dt_b_asin_title_o05_s01?ie=UTF8&psc=1

misan, 2019-08-21:
Do not forget GND for the ENBL- input.

bdring, 2019-08-21:
The ESP32 is 3.3V. Those typically use optical isolators. They require 5V (or more with external resistors). They are consume about 10mA of current. I don't think my Dev Board V3.5 will work.

Many level shifters use a pull up resistor for the high side and therfore cannot supply enough current.

I recently use a similar driver for one axis of my String Art Machine. I used simple transitors to switch the low side (PUL-) and tied the high side (PUL+) directly to 5V. It worked perfectly.

Here is part of that schematic.

[image]

I wonder what would happen if the ESP32 tried to directly sink the current on the low side

jaysettle, 2019-08-22:
It would be great to use an optotransistor here like the EL817 however I'm reading the ESP32 outputs a measly 12mA on GPIO. 

This wouldn't work for the [EL817](http://www.everlight.com/file/ProductFile/EL817.pdf)
[image]

misan, 2019-08-22:
@JAY Plus the beefier external drivers already include optoisolators at its
inputs, so adding another optoisolator on the ESP32 outputs would only
increase cost and delay over a bipolar transistor output.

jaysettle, 2019-08-22:
I missed that. Good find.

closed 2019-09-10 by bdring