bdring/Grbl_Esp32 #195: How to add support for I2C display?

Engineer2Designer, 2019-08-22:
How to add support for I2C display?

bdring, 2019-08-22:
A display is not something I am considering at this time. I feel a smart phone or tablet is far better than an I2C display.

f4grx, 2019-08-22:
There is a lot of latency with a phone, specifically with the web interface. Display is updated once per second. A hardware screen could display accurate real time machine status with much lower latency.

bdring, 2019-08-22:
I run bluetooth connections at 5 or 10 Hz.

The ESP32 has limited I/O. I prefer to reserve them for machine control.

f4grx, 2019-08-22:
Sure, that makes sense. I will have a look at the BT connections.

Engineer2Designer, 2019-08-23:
Still looking forward to it.

bdring, 2019-08-24:
I think Bluetooth or the WebUI will grow on you if you try them for a while.

I have 3D printers (Taz6 and Pulse X)E with I2C LCDs. I do like being able to use them without a computer attached. The displays are sluggish and slow to navigate, but they get the job done. The real feature is the SD card and Grbl_ESP32 has that.

One of the printers also has Octoprint. That is way better than a local LCD. Grbl_ESP32 has the ESP32 WebUI, which is very similar to Octoprint.

I am also a huge fan of GrblController for my android phone. It is so much easier to use than an LCD. It normally streams, but you can also start an SD card job from the console using [ESPxxx] commands.

Engineer2Designer, 2019-08-25:
I know that connecting via Bluetooth or other software is a great idea.If you can tell through the screen who is connected, or connected to the current situation? Or mechanical status?Can help the operator solve some problems in use.Just like the 3D printer without a screen is getting less and less.