bdring/Grbl_Esp32 #196: Made a nonworking pcb

mac7988, 2019-08-23:
I made a pcb using some tips on buildlog and can't seem to get it to work properly.
Can someone point me in the right direction. I been trying to figure out what is wrong for 2 days.( I am also dumb)=(.

https://easyeda.com/editor#id=|91c8ace7dcd94ef6801b14c2dabaed4e

misan, 2019-08-23:
@mac7988 Pololu says: If you are connecting more than 20 V or your power leads or supply has high inductance, we recommend soldering a 33 μF or larger electrolytic capacitor close to the regulator between VIN and GND. The capacitor should be rated for at least 50 V.

No capacitors shown in your PCB.

mac7988, 2019-08-23:
Thank you Misan for a quick response.currebtly I have a 9v power supply hooked up instead of 9v.

I get no movement at all from my stepper motor.
I even tried a new stepper drive and a new esp32.

I am puzzled

misan, 2019-08-23:
mac7988 I cannot know what type of wiring you did. Is it explained anywhere
else? I presume you get the USB/Web/BT working and answering properly but
again I do not really know that.

When you have a problem and need help, first let us know all details about
your setup so maybe we can help.

mac7988, 2019-08-24:
I figured it out! Turns out I had my transistor wired wrong. Fixed and working

closed 2019-08-24 by mac7988