bdring/Grbl_Esp32 #205: Recent New Features

bdring, 2019-09-03:
Quite a bit of work has been done on Grbl_ESP32 lately. Here is a list of some recent notable changes. These are all on the devt branch. They will be merged to the master branch soon.

### Grbl

**6 Axis support** You can now define up to 6 axes in Grbl_ESP32. They can be any combination of stepper motors and servos. [Here is a video of a simple machine implementing 6 axes](https://www.youtube.com/watch?v=RVi9n5H70DY). Usage information will be added to the Wiki. You can look at "CPU_MAP_FOO_6X" as an example.

**Servo Axis Updates** A few changes were made to this for easier calibration and setup in cpu_map. More setup changes are planed. The Servo axis is a really unique feature for CNC machines that allows you to control a hobby servos speed and position from gcode just like a stepper motor.

**SD Card Updates** You can now delete files from the SD card via [ESPxxx] commands. Also there is better reporting of the remaining space on the card.

### WebUI

**6 Axis Support** All area of the WebUI now support up to 6 axes. This is automatically detected. If you have axes greater than 3 the Z jog graphic on the jog panel can be changed between Z,A,B & C.

**Better Upload Management**

**Better error reporting**

**Bug fixes**

karoria, 2019-09-04:
Great News..!! I was waiting for this moment...!!

bdring, 2019-09-05:
Those changes have been merged to master

New Changes today to Devt
 - refactored kinematics to make it easy to associate different types with a different machines.
 - Servo Axis: If it detects out of range calibration values, it returns them to default values
- Added macro button feature. Use a button to home, run a job, etc.
- Added optional software debounce for control switches
- Some clean up

Currently there is a Servo Pen feature and Servo Axis feature. The Servo Axis feature does everything Servo Pen does and more. Servo Pen will be going away soon.

karoria, 2019-09-05:
Thank you Barton for this wonderful enhancement. Where can I get pinout diagram for 6 axis?  Or wiki for axis configuration. By the way, I need only 4 axis in my current project.

bdring, 2019-09-05:
The ESP32 is very flexible with pins. You can map just about any pin to any function. [See this Wiki page](https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins).

If you are creating a new design, I would look at the existing [cpu_maps](https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/cpu_map.h) and see what is closest to what you are trying to do. `CPU_MAP_FOO_6X` is a 6 axis map, but 2 of them are hobby servos.

If you are using an SD card, I would recommend the default SPI pinout that the Arduino Library uses.

[image]

karoria, 2019-09-06:
Thank you again. I got it. I am not using sd card as of now so will spare those pins for 4th axis.

bdring, 2019-09-09:
Added some demo code for forward kinematics. This is used for reporting work coordinate position.

Example: If you have a polar machine and tell it to go to X10 Y10. It coverts to radius 14.14 and radius 45°. It normally reports in those units. If you want it to report in the same system you send the gcode in and report X10 Y10, there is now an option for that....in devt branch

JackHouseLab, 2019-09-23:
Hi,
It is possible to use Flood (IO16) and Mist (IO21) pins as 4-th axe? 
Just some modification in definition files?  
What to do with 5th and 6th axis? Comment out?  
BTW: Which g-code sender for grbl can drive all 4 axis? (is there any?)

bdring, 2019-09-23:
@JackHouseLab 

Everything is defined in your cpu_map. You only need to define the features you want. If you don't define 5th (B) and 6th (C) axes they will not be used. The only catch is you must use the axes in order, so 4 axes must be XYZA.  

You can define just about any unused pins for for the 4th axis. [See this wiki page](https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins)

I have not looked into senders and axes greater than 3. The WebUI supports it, so you could use it to send code from the SD card.

JackHouseLab, 2019-09-24:
Great :-)  
I must look deeper in config files - there are many options. 
I'm using Candle as G-code sender - it has good visualisation of work and progress - it's good to know where we are ;-)
As it's open source program, so maybe someone will add more axes in the future (as fork).

closed 2020-01-10 by bdring