bdring/Grbl_Esp32 #206: Problems Compiling Firmware

Y-uXiN, 2019-09-04:
Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)?
Yes.
What version of the Arduino IDE are you using? 
1.8.9
What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using? 
v1.0.2
Are you using the master branch of Grbl_ESP32?
Yes
Have you made any edits or configuration changes (list them) to the  firmware?
Yes
Please paste the compiler error text here:
[MSG:Servo cal ($132) Error: 300.0000 s/b between 20.00 and 180.00]
[MSG:Servo cal ($102) Error: 800.0000 s/b between 20.00 and 180.00]
[MSG:Servo cal ($132) Error: 300.0000 s/b between 20.00 and 180.00]
[MSG:Servo cal ($102) Error: 800.0000 s/b between 20.00 and 180.00]
[MSG:Servo cal ($132) Error: 300.0000 s/b between 20.00 and 180.00]
[MSG:Servo cal ($102) Error: 800.0000 s/b between 20.00 and 180.00]

I have followed the step 1&2 on https://github.com/bdring/Grbl_Esp32/wiki/Servo-Axis-Feature
and made some modifications in the config.h and cpu_map.h
[config.txt](https://github.com/bdring/Grbl_Esp32/files/3573480/config.txt)
[cpu_map.txt](https://github.com/bdring/Grbl_Esp32/files/3573485/cpu_map.txt)

bdring, 2019-09-04:
That is not a compiler error. You are getting warnings because the servo calibration values for $102 and $132 are out of range. Please send $102=100 and $132=100 via serial console to put those values into the proper range.

Servos use those values to calibrate the end points of travel. The numbers are like a percentage. $102=100 would set the low end of the travel of the Z axis to 100%. 100% is essentially no calibration applied. If you set it $102=110. It would multiply the the pulse length of the lower end of travel by 1.10.

To make sure the calibration is not excessive, the range is limited to 20% to 180%.

Y-uXiN, 2019-09-04:
It perfectly solves the errors. Thank you very much.

closed 2019-09-04 by Y-uXiN