bdring/Grbl_Esp32 #213: No Configuration Panel

NickHead, 2019-09-10:
Hi all, Compiled the ptogram, uploaded it, the GRBL Configuration Panel does not show any information.
Am I doing something wrong ?
Compiled with PlatformIO and VSCode.
Nothing changed in program, all standard.
Help Please.

bdring, 2019-09-10:
Send a screen shot showing the panel.  

Did you click the refresh icon? Open a serial terminal and send $$ plus carriage return to the the ESP32. Do you get a response?

NickHead, 2019-09-10:
Tried everything. Even re-flashed quite a few times after re-compiling. It just shows Value,Name and Help but nothing underneath. 
If I do a $$ in the console, the GRBL configuration shows, but not on the html page. I am at a dead loss

bdring, 2019-09-10:
I look into it. Maybe a recent change broke something.

luc-github, 2019-09-10:
I hhave tested master and it is fine :
[image]

luc-github, 2019-09-10:
@NickHead what version do you use?

bdring, 2019-09-10:
Are you connected to hardware or is this a loose ESP32?

bdring, 2019-09-10:
This is just a guess, but the default configuration is for the controller shown on the [readme page](https://github.com/bdring/Grbl_Esp32/blob/master/README.md). It uses some I/O that requires pullup resistors. If those pins float, it could put the controller into a locked mode.

Try uncommenting `#define IGNORE_CONTROL_PINS` in config.h and recompiling the firmware.

NickHead, 2019-09-10:
Sorry about not replying fast, I am in the middle of a multitude of things. 
Thanks for all the replies though, it's really appreciated>
I downloaded the latest version of GRBLESP.
It's running on an ESP32 WROOM .
It is not connected to any hardware, just the micro, by itself. I never thought it would make a difference.
I will try the ignore control as soon as I can and report back.
Regards.

NickHead, 2019-09-10:
[E][Parsing.cpp:82] _parseRequest(): Invalid request: ␖␃␁
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[E][WiFiClient.cpp:365] write(): fail on fd 61, errno: 104, "Connection reset by peer"

That did not work either, I may try a compile on a different computer and see what happens

NickHead, 2019-09-10:
[image]

bdring, 2019-09-10:
Does the rest of the WebUI appear to work? I could send a firmware to load via OTA update.

NickHead, 2019-09-10:
Yes, the rest does look OK , the controller shows values X,Y positions. It's just the GRBL config I am having issues with and to be fair, I don't know if I am doing anything wrong.

bdring, 2019-09-10:
Here is a compressed .bin file that works for me.  Unzip it and use the yellow cloud icon on the ESP3D page to upload the .bin file.

I am out for a few hours, but can check in later.

[Grbl_Esp32.ino.zip](https://github.com/bdring/Grbl_Esp32/files/3598060/Grbl_Esp32.ino.zip)

NickHead, 2019-09-11:
Something must be wrong. Unzipped, pressed yellow cloud, accepted to update, came back with an error 0.
I am no novice to all this either , so you can be assured I know what I am doing to an extent.
Unless I have a defective ESP , I may have to see if I have any more laying about.

luc-github, 2019-09-11:
back to basic - what version do you use ?
Check the screen I sent before on top of menu, it have the versions on it, do a screenshot
[image]

also :
what  OS / browser do you use ? 
what version of esp 32 core do you use ?
you can do a screenshot of window that apear when pushing blue button :
[image]

NickHead, 2019-09-11:
Hi luc
Cant do a screen shot as it's running on a separate laptop.
Best I can do is:
UI:2 1b50 / FW: 1.1f
(20190905)

luc-github, 2019-09-11:
this is not the master branch you use the latest version of webUI which is not validated yet with GRBL
please use the indext.html.gz that is present in your grbl_esp32 fw directory in data directory 

Every version of webUI provided with FW are working - the others may have issue

NickHead, 2019-09-11:
Right, thanks, that may be where I am going wrong as I did download the index gz file externally, it was mentioned to do that in some random setup website that I read
Hopefully it will work.

luc-github, 2019-09-11:
let me know if not - we can continue to dig in issue

NickHead, 2019-09-11:
Thanks luc.
After a problem getting the wifi to work (I used arduino libraries in the compiler and it didn't like it for some reason) I have managed to get the configuration html up by sending over the correct file from the directory.
I do have an error though, but I can live with this for now.
I have always had this error from the very beginning, but won't bother you anymore (for now :) )

[E][WebServer.cpp:602] _handleRequest(): request handler not found

Just for information.

Many thanks for the help, let the fun begin.

Regards.

luc-github, 2019-09-11:
well this part is not normal and it is in FW
can you screenshot the content of blue button ?

NickHead, 2019-09-11:
When I do a refresh of the main page (F5) I get this:
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[E][Parsing.cpp:275] _parseArguments(): arg missing value: 0
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[E][Parsing.cpp:275] _parseArguments(): arg missing value: 0
[E][WebServer.cpp:602] _handleRequest(): request handler not found

NickHead, 2019-09-11:
this is monitoring on the terminal of the compiler
Not on the screen on the html page

luc-github, 2019-09-11:
yes - but  I need info on what core and configuration you are - so I need what info show in in fo window of blue button

NickHead, 2019-09-11:
UI: 2.1b39 / FW: 1.1f

I cant capture the screen as I am on a different laptop and it is on a different IP, the config from the info is above if that is any use

NickHead, 2019-09-11:
I managed a version to send
[image]

luc-github, 2019-09-11:
no what I need is the display of this blue button : 
[image]

NickHead, 2019-09-11:
ok, sorry there were too many blue buttons to contend with, 2 secs

luc-github, 2019-09-11:
because what you show it is an old version - so I am quite sure the problem is not in FW but in the way you build it so I need to know the esp32 core version

NickHead, 2019-09-11:
[image]

NickHead, 2019-09-11:
I have it now connected to a diode laser and it's off like the clappers

luc-github, 2019-09-11:
ok - looks like you use very old version of core and old fw
1 - please download latest master here : https://github.com/bdring/Grbl_Esp32/archive/master.zip
2 - change core version in platformIO.ini to use latest git version
`platform = espressif32`
by 
`platform = https://github.com/platformio/platform-espressif32.git`

you should have all latest version

luc-github, 2019-09-11:
once done you must have this output :
[image]
You must not be in mixed mode like you are now - if yes restart board it should be in station only

NickHead, 2019-09-11:
OK I will try that as well.
The board is an Expressif Devkit V1 using the WROOM chip by DoIt.
The actual compiler properties I used were:

 [env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1

Which I think are correct for this board.

if you say your above board details will be batter, then I will try them out, thanks.

luc-github, 2019-09-11:
stable is now old - better use the latest git

NickHead, 2019-09-11:
I didn't change any modes on the wifi, only the ip address from 192.168.0.1 to 192.168.0.23 as it clashed with my router. The wifi is as set in the config/wificonfig.h files, so I don't know where the mixed mode is coming from as I don't think it was my doing.

luc-github, 2019-09-11:
Many possible reason not related to your actions

NickHead, 2019-09-11:
Yer, tried all that, re-flashed, looks like the same SDK, wifi mode still Mixed. Tried a few changes in the config file like #define DEFAULT_RADIO_MODE ESP_WIFI_STA, still all the same.
it's working in a fashion, better than it ever was.

NickHead, 2019-09-11:
same exactly - my platformio must have the most up to date info installed.
GRBL is 1.1f, but I did notice that the UI is 2.1b39 and not 1b50 as I had before, that was from the link you sent me to upload

luc-github, 2019-09-11:
Please show me your platformio.ini

NickHead, 2019-09-11:
[platformio]
src_dir=Grbl_Esp32
lib_dir=libraries
data_dir=Grbl_Esp32/data

[common_env_data]
lib_deps_builtin =
	ArduinoOTA
	BluetoothSerial
	DNSServer
	EEPROM
	ESPmDNS
	FS
	Preferences
	SD
	SPI
	SPIFFS
	Update
	WebServer
	WiFi
	WiFiClientSecure

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
upload_speed = 512000
board_build.partitions = min_spiffs.csv
monitor_speed = 115200

NickHead, 2019-09-11:
I am just updating with that OTA file you sent, it wouldnt go in before, it is now

NickHead, 2019-09-11:
SDK now updated to v3.2.2.120
wifi Mode still mixed
UI 2.1b39 / FK 1.1f  20190908

luc-github, 2019-09-11:
What is the output of `[ESP400]` command?

NickHead, 2019-09-11:
{"EEPROM":[
{"F":"network","P":"ESP_HOSTNAME","T":"S","V":"grblesp","H":"Hostname" ,"S":"32", "M":"1"},
{"F":"network","P":"HTTP_ON","T":"B","V":"1","H":"HTTP protocol","O":[{"Enabled":"1"},{"Disabled":"0"}]},
{"F":"network","P":"HTTP_PORT","T":"I","V":"80","H":"HTTP Port","S":"65001","M":"1"},
{"F":"network","P":"TELNET_ON","T":"B","V":"1","H":"Telnet protocol","O":[{"Enabled":"1"},{"Disabled":"0"}]},
{"F":"network","P":"TELNET_PORT","T":"I","V":"23","H":"Telnet Port","S":"65001","M":"1"},
{"F":"network","P":"RADIO_MODE","T":"B","V":"2","H":"Radio mode","O":[{"None":"0"},{"STA":"1"},{"AP":"2"},{"BT":"3"}]},
{"F":"network","P":"STA_SSID","T":"S","V":"GRBL_ESP","S":"32","H":"Station SSID","M":"1"},
{"F":"network","P":"STA_PWD","T":"S","V":"********","S":"64","H":"Station Password","M":"8"},
{"F":"network","P":"STA_IP_MODE","T":"B","V":"0","H":"Station IP Mode","O":[{"DHCP":"0"},{"Static":"1"}]},
{"F":"network","P":"STA_IP","T":"A","V":"0.0.0.0","H":"Station Static IP"},
{"F":"network","P":"STA_GW","T":"A","V":"0.0.0.0","H":"Station Static Gateway"},
{"F":"network","P":"STA_MK","T":"A","V":"0.0.0.0","H":"Station Static Mask"},
{"F":"network","P":"AP_SSID","T":"S","V":"GRBL_ESP","S":"32","H":"AP SSID","M":"1"},
{"F":"network","P":"AP_PWD","T":"S","V":"********","S":"64","H":"AP Password","M":"8"},
{"F":"network","P":"AP_IP","T":"A","V":"192.168.0.1","H":"AP Static IP"},
{"F":"network","P":"AP_CHANNEL","T":"B","V":"1","H":"AP Channel","O":[{"1":"1"},{"2":"2"},{"3":"3"},{"4":"4"},{"5":"5"},{"6":"6"},{"7":"7"},{"8":"8"},{"9":"9"},{"10":"10"},{"11":"11"},{"12":"12"},{"13":"13"},{"14":"14"}]},{"F":"network","P":"NOTIF_TYPE","T":"B","V":"0","H":"Notification type","O":[{"None":"0"},{"Line":"3"},{"Pushover":"1"},{"Email":"2"}]},
{"F":"network","P":"NOTIF_T1","T":"S","V":"","S":"63","H":"Notification Token 1","M":"0"},
{"F":"network","P":"NOTIF_T2","T":"S","V":"","S":"63","H":"Notification Token 2","M":"0"},
{"F":"network","P":"NOTIF_TS","T":"S","V":"","S":"127","H":"Notification Settings","M":"0"}
]}

luc-github, 2019-09-11:
Are you using your device in AP mode ? How did you setup the wifi for your grbl ?

NickHead, 2019-09-11:
i didnt, it's all standard. downloaded a fresh version , compiled and run it, wifi is just on as esp, connected using the ip and thats it, using lightburn via comport to run the laser, wifi just monitors the output and can obviousley move the laser about using jog .

luc-github, 2019-09-11:
Is not recommended to use in AP mode. AP mode is normaly to let you configure your network. Your pc connected to esp will send lot of queries thinking they are on internet
Do you still have the error in terminal when you refresh page ?

NickHead, 2019-09-11:
Yes I do, I am effectively only monitoring the USB output with either Lightburn, or Platformio Terminal.
It still works, but I am getting that error. No big deal really.

NickHead, 2019-09-14:
Managed to get the wifi out of Mixed Mode
Had an unused variable in the 12c hal C file so edited that out.
editied an undeclared variable in the stepper.h (RMT or summat or other) that caused a  flag up.
All seems good now, no errors on compile.
found out that I had to erase the ESP32 first for it to accept the new wifi settings.
[image]
Many thanks for all the help, I am sure there will be more to come.
regards, Nick

bdring, 2019-09-14:
@NickHead 

Are you saying your changes fixed it? @luc-github made some changes too. 

What variable was undeclared?

luc-github, 2019-09-14:
Now he is in station mode so no mixed mode issue

closed 2019-09-14 by bdring

NickHead, 2019-09-15:
The errors I got were from the I2C hal. Probably in the compiler in the Platformio that I have installed.
It was a variable that was declared and never used. wasn't an error as such, it just flagged up everytime I compiled, so i remove it from the hal routine.
Also in the stepper.h file, another one that mentioned RMT, I can't remember fully, if you do require it, I can replace it, re-compile and send you the info.
They don't cause a problem, just annoy me when they show up.

NickHead, 2019-09-15:
I re-did it for you.
Here is number 1.

Grbl_Esp32\stepper.h:86:35: warning: 'void stepperRMT_Outputs()' declared 'static' but never defined [-Wunused-function]
      inline IRAM_ATTR static void stepperRMT_Outputs();  //removed this line as it caused errors on compile 
                                   ^
In file included from Grbl_Esp32\grbl.h:56:0,
                 from Grbl_Esp32\spindle_control.h:24,
                 from Grbl_Esp32\grbl.h:55,
                 from Grbl_Esp32\serial.h:24,
                 from Grbl_Esp32\grbl.h:54,
                 from Grbl_Esp32\report.h:24,
                 from Grbl_Esp32\grbl.h:53,
                 from Grbl_Esp32\motion_control.h:30,
                 from Grbl_Esp32\grbl.h:49,
                 from Grbl_Esp32\grbl_eeprom.h:24,
                 from Grbl_Esp32\grbl.h:46,
                 from Grbl_Esp32\system.h:23,
                 from Grbl_Esp32\grbl.h:42,
                 from Grbl_Esp32\settings.h:28,
                 from Grbl_Esp32\grbl.h:41,
                 from Grbl_Esp32\tdef.h:4,
                 from Grbl_Esp32\grbl.h:38,
                 from Grbl_Esp32\notifications_service.cpp:33:

luc-github, 2019-09-15:
They are warning, not errors
You can ignore them
PlatformIO default compilation warning are more strong than arduino ones

NickHead, 2019-09-15:
Here is the other one:

Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\esp32-hal-i2c.c.o
C:\users\nova\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-i2c.c: In function 'i2cProcQueue':
C:\users\nova\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-i2c.c:1282:18: warning: unused variable 'tAfter' [-Wunused-variable]
     portTickType tAfter=xTaskGetTickCount();
                  ^
C:\users\nova\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-i2c.c:1274:18: warning: unused variable 'tBefore' [-Wunused-variable]
     portTickType tBefore=xTaskGetTickCount();
                  ^

NickHead, 2019-09-15:
I don't like to see them in the console so I removed them.

mrarmyant, 2020-03-17:
> i didnt, it's all standard. downloaded a fresh version , compiled and run it, wifi is just on as esp, connected using the ip and thats it, using lightburn via comport to run the laser, wifi just monitors the output and can obviousley move the laser about using jog .

How did you set that comport up?  I'm trying to do the same thing.

NickHead, 2020-03-17:
I can't remember now, I think I just plugged it into the ESP, I have one here that I just plug in to the serial.
For serial over IP I use Tibbo , But I am only using that for RS485 protocol on a totally different area , not lasers.