bdring/Grbl_Esp32 #216: Test Drive Mode

bdring, 2019-09-11:
I created a new CPU_MAP called CPU_MAP_TEST_DRIVE. This creates a virtual 3 axis machine with no actual i/o mapped.

This allows you to safely test Grbl_ESP32 on an ESP32 module that is not attached to a carrier board or is attached to some unknown hardware. This removes problems with floating input pins or possibly grounded  output pins. You will not get erroneous readings on inputs or damage grounded outputs. It also does not require any 3rd party libraries installed. 

Of course, you will not be able to do things like home.  It will just give you the normal errors if you tried to home without limit switches.

You can safely play around and test drive the firmware. It also sets up the partitions and wifi, so future uploads could be done via OTA (over the air) firmware uploads.

I was thinking about making this the default CPU_MAP so true newbies have the best chance of success without hurting anything

Thoughts?

luc-github, 2019-09-12:
Very good

misan, 2019-09-12:
The problem I see, if this becomes the default, is that a newbie may next complain about his hardware not moving at all even though the interface shows the move ;-)

closed 2019-09-14 by bdring