bdring/Grbl_Esp32 #217: Trying to make a pcb need advice on my method

mac7988, 2019-09-12:


mac7988, 2019-09-13:
@bart, I see my mistake with the input circuit. You are right it will short the entire system. How do I know? It did short the entire system.😑😒 
Lucky it's only a breadboard trial.

@karoria I will change my step output voltage to 5 volts. My reason to move to 24volt is the ability to use industrial limit sensor for accuracy and reliability.

karoria, 2019-09-13:
Oh. That is the exact case of mine. I want to share you my experience that most of the inductive type limit sensors work perfectly with 5v !!!! However they mention 12v-24v operating range. I have tried with 2 different manufacturer and both of them worked nicely at 5v. So my suggestion is to buy one such sensor and test it. Yes, i have also noted that they dont work on 3.3v. FYI.

karoria, 2019-09-13:
So, I take 5v as input and the output signal which is also 5v has to be dropped down to less than 3.3v by using resistor. Value of resistor will depend on the operating current of your sensor.

mac7988, 2019-09-13:
@karoria if I use a optocoupler than I don't have to work a out it as there is no electrical connection.

karoria, 2019-09-14:
I tried all the way. The problem with opto is response time and so the distorted waveform again. But i am not discouraging you as in my case also correct resistors given at least more than 18v step pulse though not 24v. With wrong resistor value you can end up with merely 5v to 8v signal. You have to watch output in DSO to understand what I mean. 
As I already told, if your only concern is to take input from industrial sensor, go for it with 5v circuit. I have got consistent results so I recommend that way.

mac7988, 2019-09-14:
Hi Ravi,

Please check the current setup. I don't have a opto on the output so I
don't have an issue here.

karoria, 2019-09-14:
Give it a try on a breadboard first. You need to attach a resistor before ground. This way you will get non-inverting pulse at collector output.