bdring/Grbl_Esp32 #220: Trinamic Drivers

bdring, 2019-09-14:
Recently the TMCStepper library has been upgraded to support ESP32 better and handle daisy chaining. 

I was previously using my fork of TMCStepper for some testing of these features. Now that everything I need is in the original library, I am going to update the devt (then master) branch to use the official TMCStepper library. It was using the TMC2130Stepper library. That will bring in support for lots of drivers and daisy chaining.

I like the CPU_MAPs being one of the only things a users needs to touch to define a machine. That gets a little messy with the dozens of things you can do with these drivers. I'll start with support for the major features, like micro stepping and motor current. 

Advanced users may need to edit other files until I figure out a solution.

ewidance, 2019-09-16:
Well done! I've recently updated my ultimaker with 32 bits LPC1768 + Trinamic + Marlin 2. Trinamic are really impressive... I'm goind t buy some more to test with GRBL/ESP32...

bdring, 2019-11-02:
@alfredanil 

TMC5160 is already there. I don't have any of those other drivers and it will take some time to add the code. I'll add it to the [roadmap](https://github.com/bdring/Grbl_Esp32/wiki/Development-Roadmap).

shreeramlive, 2019-12-17:
Hi, The` TMC5160 SPI drivers` have driven me crazy enough for almost a week :relaxed: I'm not able to check the driver **SPI connection** with the ESP32. Still not able to make them move. 
Please suggest some way to **debug the SPI connection** of drivers.

bdring, 2019-12-17:
What controller board are you using?

shreeramlive, 2019-12-18:
Hi @bdring, I'm testing a custom made board. bdring/4_Axis_SPI_CNC#3

bdring, 2019-12-18:
I always use a logic analyzer I have a Logic 4 from Saleae.

shreeramlive, 2019-12-21:
Thanks for the suggestion.
I don't have any Logic analyzer yet:satisfied:, Anyway I will try my best to make it work.

shreeramlive, 2019-12-30:
Hi, After searching about those **TMC5160** drivers. I realized that the `TMC5160 stepsticks` have their internal `motion control` disabled since the `STEP/DIR` interface is used.

These drivers have the ability to be controlled with `SPI`alone using their `integrated motion controller`, this can be beneficial in **freeing up some pins** on `ESP32`.
http://blog.trinamic.com/2017/04/05/how-to-use-tmc5130-eval-with-your-arduino-mega/
http://blog.trinamic.com/2018/02/19/stepper-motor-with-tmc5160/

Moreover the `simple encoder interface` on TMC5160/TMC5130 can be used to get positions directly from the driver's SPI itself. So `closed loop` control is also a possibility since ESP32 has good processing power.
@uStepper's [**uStepperS** ](https://github.com/uStepper/uStepperS/blob/master/Hardware/Schematics.PDF)board uses TMC5130 with an encoder interface for closed loop control.
https://github.com/uStepper/uStepperS

So this kind of control would be beneficial for large **CNC mills, Laser cutters, etc**. This probably can be an awesome choice where StepSticks are a no go. It can help to `reduce the cost` of the electronics ( ESP32 has already reduced the cost a lot by removing the need for a host ).

Resources:

- A library created by @tommag https://github.com/tommag/Estee_TMC5130
- Trinamic guide: [Adapting step/dir CNC firmware for ramp generators ](https://www.trinamic.com/fileadmin/assets/Support/Appnotes/AN046-Adapting_step_dir_cnc_firmware_for_ramp_generators.pdf)

Concerns:
- **SPI** is also shared by the `SD card`, so is it possible for this type of control with hindering the SD card features. 
- [Teemuatlut's TMCStepper](https://github.com/teemuatlut/TMCStepper) is nice for **configuration** of Trinamic drivers but it doesn't support motion control feature. 
- This may not be cost-effective for applications like small laser engravers. Highly possible that majority of users may not need this feature.

Please share your opinions!

bdring, 2019-12-30:
This has been on my mind for a while now. I have even discussed this with Sonny, the developer of Grbl.

It would be dozens of hours of work and require some expensive custom designed hardware.

Right now the SPI is shared with the SD card. This has no penalty because the stepper driver SPI commands are only used at startup or in idle. 

Daisy chain mode might also need to be tested to see if it is viable. 

The current TMCStepper library probably would need some major updates.

Ramp vs step/dir. This would require a major rewrite of Grbl. There may be some edge cases where there are problems, like Grbls realtime overrides and fast freehold.

Closed loop. I think this would need to be done in Grbl with step and dir or direct PWM control.

Encoders. It is possible encoders could be used for verifying no steps are lost and allowing manual moves (pushing axes) to be allowed and position updated.

Bottom line. I am not ready to spend so much time and money on this experiment at this time.

closed 2020-01-10 by bdring