bdring/Grbl_Esp32 #222: Can't upload firmware depending to what's connected to D2 (MPCNCV1P2 Pinout)

ewidance, 2019-09-15:
Hi,

I've wired a prototype with  MPCNCV1P2 pinout. I used level shifter modules with KSA1298 transistors (CMS marking J18). On the low V part of modules, there are 10K pullups. 

After wiring i could not flash anymore via serial. Since my ESP module was solederd on my own shlied, i could not detach it. I searched a lot to find what wass messing the prog function and found. 

It was because off the 10K resistor pullup on D2. I used instead a transistor basically wired (PNP) with success since is was not needing pullup. Flashing via serial was back.

I hope this could help somebody else, and maybe this could be put on "programming GRBL" wiki page : Avoid pullups on D2 and don't wire modules you don't kow how they are cabled..

Sincerely,

JP CIVADE

bdring, 2019-09-15:
I assume you mean GPIO2. There is some documentation on that at [Espressif](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection). I can add that to the ["Setting up the i/o pins page"](https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins)

bdring, 2019-09-15:
The pages have been edited.

closed 2019-09-15 by bdring