bdring/Grbl_Esp32 #223: Laser mode don't work as expected ($32=1)

ewidance, 2019-09-15:
I've wired a prototy using MPCNCV1P2 pinount. The firmware is compiled with  CPU_MAP_MPCNC_V1P2, In cpu Map, USE_SPINDLE_RELAY is commented, and COOLANT_FLOOD_PIN  is set to  GIO_NUM_2.

Laser SPINDLE_PWM_PIN is on GPIO_NUM_16 (with a level shifter for 5V), SPINDLE_ENABLE_PIN is on 32 (not used), and COOLANT_FLOOD_PIN on pin2 is wired to a relay for Laser Fan. 

When laser mode is off ($32=0), Laser power is properly controlled with M3 S0 > 255/M5. Fan is working (M8/M9). Spindle enable pin don't seems to work (i've pu a led on it). 

Since laser mode is supposed only to allow instant PWM changes without wating for a spindle to rise up, i've tried to set it on ($32=1).

Laser fan (coolant flood) continue working ok, but the's not anymore PWM on SPINDLE_PWM_PIN.

In both cases, SPINDLE_ENABLE_PIN  stays off.

Now the questions :
* Why SPINDLE_PWM_PIN is not working in Laser mode?
* Why SPINDLE_ENABLE_PIN   stays always off?

Thanks you for you time and patience, Bart. Again a big thank you for the job done...

JPC

bdring, 2019-09-15:
I'll look into your problem on that controller this week.

bdring, 2019-09-16:
It seems to work for me. The in laser mode, spindle enable is always on during M3 and M4 and off during M5. I am using the stock CPU_MAP_MPCNC_V1P2 cpu_map.

In laser mode...

M3 works very similar to non laser mode, except S values are immediately applied to the PWM.

M4 compensates for the exact speed compared to desired feed rate during accel and decel. Therefore it goes to zero after a move is done.

Here is some M3 gcode that works for me. Notice all the S values. This was generated by laser Grbl.
```
G90
G0 X40 Y34
M3 S0
F1000
G1 X55.333 S255
G0 X55.667 Y34 S0
G1 X118.333 S255
S0
G0 X118.333 Y34.333 S0
```

bdring, 2019-09-16:
Can you paste the minimal amount of gcode that shows the problem you are having?

ewidance, 2019-09-17:
I've tested with your Gcode. WIth laser mode off, laser powers on. WIth laser mode on, laser don't power on. I've put a led on pwm output to appreciate modulation.

Just in case, here's my $$ config:
```
$0=3
$1=250
$2=0
$3=0
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=3
$24=600.000
$25=2000.000
$26=250
$27=1.500
$30=1000.000
$31=0.000
$32=1
$100=1000.000
$101=1000.000
$102=1000.000
$110=6000.000
$111=6000.000
$112=3000.000
$120=1000.000
$121=1000.000
$122=1000.000
$130=500.000
$131=500.000
$132=80.000  
```

As asked, here's a minimal amount of code generated with lightburn:

```
; LightBurn 0.9.07
; GRBL-M3 (1.1e or earlier) device profile, absolute coords
G00 G17 G40 G21 G54
G90
; Cut @ 6000 mm/min, 100% power
M8
M5
G0X6Y5
M3
G1Y20S255F6000
G1X33
G1Y5
G1X6
M5
M9
G1S0
G90
; return to user-defined finish pos
G0 X0 Y0
M2
```

Same results.. No output when $32=1...

I really don't see whats happening.

ewidance, 2019-09-17:
Version infos:
UI: 2.1b51 / FW: 1.1f (20190905)

bdring, 2019-09-17:
I might have found the problem at around [line 277](https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/stepper.cpp#L277), in stepper.cpp, look for this code

```C++
#ifdef VARIABLE_SPINDLE
	// Set real-time spindle output as segment is loaded, just prior to the first step.
	if (st_prep_block->is_pwm_rate_adjusted) {
		spindle_set_speed(st.exec_segment->spindle_pwm);
	}
#endif
```
Change it to...

```C++
#ifdef VARIABLE_SPINDLE
	// Set real-time spindle output as segment is loaded, just prior to the first step.			
	spindle_set_speed(st.exec_segment->spindle_pwm);
#endif
```

compile and upload firmware

ewidance, 2019-09-17:
Gotcha! Works fine now in both modes on pwm pin. Well done :) ... Should SPINDLE_ENABLE_PIN be activated during laser fire?

bdring, 2019-09-17:
For me...

M3 is SPINDLE_ENABLE_PIN on
M5 is SPINDLE_ENABLE_PIN off. 

It does not not depend on PWM value. PWM 0 still has enable on during M3

ewidance, 2019-09-17:
Don't works. I've tested with SPINDLE_ENABLE_OFF_WITH_ZERO_SPEED defined (in case M3S0). Same thing. I'll check the wiring tomorrow. It's 11h30 PM in france. Thanks a lot for the great support. I'll post results on my tests.

bdring, 2019-09-17:
Are you testing in the MPCNC V1P2 controller? If so, is power connected to the controller or just the USB power?

ewidance, 2019-09-18:
No, as said inthe beginnning of thread, "I've wired a prototype using MPCNCV1P2 pinount". It's not a MPCNCV1P2 (there where no more available for sale). But yes i'm testing with both powers.

ewidance, 2019-09-21:
I've double check wiring, and the level shifter on GPIO32 (Spindle enable). After reparing, SPINDLE_ENABLE Works as expected. Bug ca be closed, and commit done on modification you've dont on stepper.cpp. Thank's again!

closed 2019-09-21 by bdring