bdring/Grbl_Esp32 #225: How to check for hard limit at initialization

karoria, 2019-09-16:
I am using NO type limit switches. When i get hard limit triggered, I want the machine not to be unlocked manually. It should only accept homing command in that condition, that too with first releasing off the triggered limit switch and then homing. This is because in unlocked state there are high chances for wrong movement and physical collission.
Present scenario is just checking whether limit switch is triggered or not but it lets you unlock also. Upon unlocking, the Alarm should ideally go off only when the limit switch trigger is lifted off. 
Presently machine just registers whether the limit switch state is changed or not so $5 setting doesnt have actual meaning as I see.
Am I missing something important in config or somewhere? Or this behaviour is accepted and we have to be used to it?
Waiting for help...

bdring, 2019-09-16:
- Are you using soft limits as well as hard limits?
  - Are you using switches at the min and max of the axes you are talking about?

karoria, 2019-09-16:
1. Soft limits are disabled.
2. Still I am checking at breadboard level, not fitted to machine. So min or max doesn't matter. 

I made progress. Now, at least i am able to do it by changing some config options. I think i will make it from now. The only thing I wonder is the wait time for first bounce is too low before firing an alarm of incomplete homing. How can I change it?

bdring, 2019-09-16:
#define DEBOUNCE_PERIOD 32

In config.h. 

You need to enable the line before it too.

karoria, 2019-09-16:
Ok. Got it. Will try and revert back. BTW, 32 is milliseconds?

bdring, 2019-09-16:
Yes, milliseconds.  That is mentioned in the comment following the define.

```
//#define ENABLE_SOFTWARE_DEBOUNCE // Default disabled. Uncomment to enable.
#define DEBOUNCE_PERIOD 32 // in milliseconds default 32 microseconds
```

bdring, 2019-09-16:
Notice the comments associated with `#define LIMITS_TWO_SWITCHES_ON_AXES` in config.h

That may answer some of your questions.

karoria, 2019-09-17:
Thanks. Actually I got it worked with that only. Will update when i test it.

closed 2019-09-17 by bdring