bdring/Grbl_Esp32 #233: ESPxx commands for SD Delete, Upload, Create Directory

openbuilds-engineer, 2019-09-26:
So, i have a feature request.
Apart from some cool web based gcode generator work, I am also adding support for ESP32Grbl to my own host application.
Now as-is I have awesome connectivity over serial and telnet working.
But I noticed some functionality seems to be only available over http
For example:
SD Delete
SD Upload
SD Create Dir
etc
Other functionality over HTTP uses the same command set as serial/telnet so from coding a host perspective its easier to have a function that looks at "how we are connected" and then send the same command, to the relevant interface, than having to deal with very different ways of interface, or even worse, things available on only one of the three interfaces.
So, how can we go about adding some extra [ESP* commands for things like
- Upload SD (Reprap way maybe, ESPxx1 /path/subfolder/FILENAME.GCODE starts upload, we stream the gcode, get OK for each line back so i can use existing flow, send ESPxx2 when upload is done for example)
- Delete from SD (ESPxxx3 PATH/FILENAME.GCODE for example)
- Create Directory etc

mac7988, 2019-11-11:
I would also like a way to upload directly into sd card using serial. This is ideal for a headless machine. I did realize that Marlin does this too. That being said it also has to many little issues that over complicate the firmware. I like the robustness of grbl and think it's missing two features that would make it really awesome. LCD support and SD card write support.

openbuilds-engineer, 2019-11-11:
Just an update, havent forgotten about this, but the rental we were staying in got sold from under us, so  I bought a fixer upper and did a reno the last month and a bit. Almost done! When I am settled I hope to tackle this first thing!

mac7988, 2019-11-11:
> Just an update, havent forgotten about this, but the rental we were staying in got sold from under us, so I bought a fixer upper and did a reno the last month and a bit. Almost done! When I am settled I hope to tackle this first thing!

Are you guys in Jersey?