bdring/Grbl_Esp32 #241: Add g-code command to control the relay

KrisKelvin1, 2019-10-03:
how to add a command to control an external relay in the likeness of M7, M8 assigning a free processor output.
I think that you can use the M106 command to turn on and off the M107

bdring, 2019-10-03:
Can you use M7, M8, and M9?

KrisKelvin1, 2019-10-03:
yes, I use M7 and M8 but I want to have three, and better four channels of control of external devices

mac7988, 2019-10-04:
I would love more M codes as well maybe a sub I/O board would be a great way.

bdring, 2019-10-08:
This is a common request, but I have been reluctant to add commands that are not in regular Grbl. I added it to the roadmap, but probably won't work on it for a while.

KrisKelvin1, 2019-10-08:
very sorry .... many users wanted to be able to control external devices :(

bdring, 2019-10-09:
Should the format be [M106](https://reprap.org/wiki/G-code#M106:_Fan_On) Px Sxxx. With the P value the pin and the S value the PWM? 
M107 would turn all defined pins off?

I think it would be best to to use virtual I/O pin numbers defined in a CPU map like...

`#define USER_IO_PIN_1 GPIO_NUM_21`

What is a reasonable number of pins to support?

Should any of the other parameters values be supported? Note: Grbl only supports these letters F, I, J, K, L, N, P, R, S, T, X, Y, Z
Should the pins behave in a special way if the Grbl is reset, has alarm or error?

terjeio, 2019-10-09:
AFAIK grbl tries to follow [linuxcnc specifications](http://linuxcnc.org/docs/2.5/html/gcode/m-code.html#sec:M62-M65) where relevant.

bdring, 2019-10-10:
I agree with the LinuxCNC comment

They would be [M62, M65, M64 and M67](http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m62-m65)

There is no mention of the S parameter, but we could default to full on/off if that parameter is not supplied.

bdring, 2019-10-10:
Looking into this further. It is probably best to just support the synchronized versions initially.

[M62, M63](http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m62-m65) for digital
[M67 ](http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m67)for analog

bdring, 2019-10-13:
I think I have M62 and M63 working. I need to test and think a while before I push the changes.

The format is M62 P1 to turn on pin 1 and M63 P1 to turn it off. The pin number does not refer to the actual ESP32 pin number. It is the IO you assign to be pin 1. It is done in the cpu map like this.

```C++
#define USER_DIGITAL_PIN_1	GPIO_NUM_21
#define USER_DIGITAL_PIN_2	GPIO_NUM_25
```

You can define up to 4 right now.

The command is synchronized, so it waits for all previous moves to complete before it changes the pin state. 

```
G0 X0
G1 X500 F100
M62 P1
G1 X1000
```
 
This will move quickly to X0. Slowly to X500 (decelerating to a stop). Turn on Pin 1. Move to X1000.

bdring, 2019-10-15:
[Changes](https://github.com/bdring/Grbl_Esp32/commit/515ce82672d2da2665d3c029602377599a5c161e) pushed to [devt branch](https://github.com/bdring/Grbl_Esp32/tree/Devt).

Will merge to master soon.

closed 2019-10-15 by bdring