bdring/Grbl_Esp32 #243: Estop and difference between spindle_enable and spindle_pwm

classbproject, 2019-10-07:
I'm making my own PCB since I needed two drivers for my Y-axis. I had a couple of Qs around this.

1. Is there an emergency stop input? If so, how and where would I define this? 
2. I see a spindle_enable and a spindle_pwm output. How does this work? Is there a separate gcode for turning on the pin marked as spindle_enable? In the original grbl there is just the spindle_PWM output.

bdring, 2019-10-07:
1. E stop typically cuts all power.
2. Spindle enable is optional. It is often used used with speed controllers that have an input for that. It is in original grbl.

misan, 2019-10-07:
E-stop might work better in certain setups by blocking the step and spindle
signals, so steppers stop ASAP and remain at that location (though this
does not guarantee an instant stop it should stop sooner that without
power, plus it ensures the machine stays at the last location).
Regulations in a given country might, however, call a different outcome.

closed 2019-10-08 by classbproject