bdring/Grbl_Esp32 #244: Spindle PWM invert option

owenduffy, 2019-10-08:
I would find an option to invert spindle PWM output of advantage.

Some 'break out boards' have on board PWM to analogue converters that require inverted PWM.

The feature will appeal to users of popular Chinese CNC routers that have such boards fitted (the parallel port BOBs) in avoiding a makeshift hardware inverter.

It seems that Mach3 provides such an option, certainly boards such as the TX14692 and its many variants use inverted PWM.

bdring, 2019-10-08:
I'll take a look at that later this week to see if it can be easily and cleanly added.

bdring, 2019-10-11:
I think I have this working. I am doing an overall cleanup of the spindle at the same time, so I need to do some more testing.

Previously, if you did not specify a spindle pin, you still needed to define a bunch of spindle values to get it to compile. Now, if you don't specify the pin, the other values are not needed.

owenduffy, 2019-10-11:
Ok, sounds good Bart. Happy to test when you are ready.

Please see issue #247 which relates to spindle control and user customisation... it is a very simple enhancement probably to the same module if it fits with your ideas on user customisation.

bdring, 2019-10-11:
I saw that issue. I wanted a clean PR on the invert issue before I look at that other issue. There is also the `#define ENABLE_PIECEWISE_LINEAR_SPINDLE` feature of AVR Grbl that was never implemented in Grbl_ESP32.

How are you measuring spindle speed? 

How well does it hold RPM under load?

owenduffy, 2019-10-11:
Measuring speed:

All tests are unloaded on a CNC6040 1.5kw 2pole induction motor VFD. The VFD controller displays VFD supply frequency. Induction motors have slip which is load dependent (it is the mechanism by which motor current increases to meet the load requirement), and I have verified the spindle speed unloaded reconciles with freq display using an optical tacho on the shaft (ie unloaded slip is very low as expected).

I would argue that unloaded speed is the target, as cutter burning on light cuts is related to near unloaded speed. In this context, I think you suffer loss of speed under load as an essential characteristic of the motor.

This particular VFD is not very stable unloaded at low rpm, and I expect even less so when loaded. There are practical limits to speed control.

I expect that slip at high speed is probably small, but at the lowest frequency / speed where winding reactance is lower relative to winding resistance, the slip curve is probably less ideal. Did I say there are practical limits to speed control?

The exit idea I proposed does not limit the compensation to a piece wise one, though I saw the warning to keep the code path length short.

bdring, 2019-10-11:
The changes have been uploaded to the devt branch. I will leave them there for a while before merging with master.

Any testing you can do or feedback you have is appreciated.

owenduffy, 2019-10-11:
I should be able to do that today (UTC+11).

owenduffy, 2019-10-11:
I have tested it initially with a scope on the output.

Verified that it is inverted, but there is an issue at speeds below $31 speed in that I expected PWM of 0% (inverted) but it is providing 1.56µs (I assume PWM=1/256) and we really want PWM=0 when speed is below S31 don't we?

If I set speed S2000, I get 10.15µs (inverted fine), and them M2 and PWM=770ns, we need 0ns to turn the motor off.

Have I been clear?

bdring, 2019-10-11:
I didn't test that. I never liked the way that works. It seems like it is not very robust.

Is the purpose of that to make sure you run the spindle at a minimum RPM or a minimum PWM?

I think it should just change any non zero Sxxx below your minimum to the minimum and send it through the same calcs as any other S value. Otherwise, unless SPINDLE_PWM_MIN_VALUE is perfect you have am odd transition from below minimum to the minimum.

Also

I think S should be tested and fixed in gcode.cpp. If you you send S100 it would change it to S200 (if that is the min). Now the $G reporting and overrides, etc work on the right value.

owenduffy, 2019-10-11:
Ok, I think the system needs to send PWM 0% (non-inverted or inverted as per config) whenever the motor is stopped, program is stopped, or Sxxxx<min. It should be done in a reliable and consistent way.

So, for the inverted case, I expect to see 3V steady on the spindle pin when any of the conditions in par 1 apply.

The reason I did not wade into making the change myself is lack of knowledge of the interaction of all the various modules and parameters (the spaghetti if you like).

Happy to test again when you are ready, again I would use the scope first.

Thanks for your effort Bart, appreciated.

BTW, I could live with the grbl linearisation... live with it???, it would be much better than none and is essentially the same as I implemented and tested.

Well done!

owenduffy, 2019-10-11:
I might explain that the reason I think the spindle should be stopped for sxxxx<rpmmin is that it might not be safe to spin the 'tool' at rpmmin, so it should not be a default speed. An example would be an edge finder that might not be safe above 600rpm.

On the other hand, capping sxxx at rpmmax makes sense.

bdring, 2019-10-11:
I am not sure how it saves the edge finder if you set the RPM too high.

I suppose if you are really worried about all this stuff, you will set it up correctly. Maybe I'll mostly leave it alone.

How about this...

Example:
settings.rpm_min = 200
settings.rpm_max = 1000

S0 = 0 PWM
S100 = SPINDLE_PWM_MIN_VALUE
S200 = 200 RPM
S1000 = 1000 RPM
S1200 = 1000 RPM (and reports 1000)

owenduffy, 2019-10-11:
Let me try to say it more clearly.

The system should send PWM=0% (inverted or not as configured) for:
-Sxxx<rpm_min;
-at startup;
-whenever the program is stopped (M2).

Reasoning: it may not be safe for motor or driver to command it too slow, so if the user properly configures rpm_min ($31), the code should protect the motor if lower speed is requested.

(The edge finder is a digression, excessive speed may damage it or the work. So if I order s600 and the system is configured for rpm_min=1200, send PWM=0%.)

owenduffy, 2019-10-11:
> I am not sure how it saves the edge finder if you set the RPM too high.
>
> I suppose if you are really worried about all this stuff, you will set it up correctly. Maybe I'll mostly leave it alone. [...]

S100 = 0 PWM

bdring, 2019-10-11:
I disagree.  

Reasoning:

Never accept an RPM but not spin.

Here is "original" Grbl.

```
else if (rpm <= settings.rpm_min) {
        if (rpm == 0.0) { // S0 disables spindle
          sys.spindle_speed = 0.0;
          pwm_value = SPINDLE_PWM_OFF_VALUE;
        } else { // Set minimum PWM output
          sys.spindle_speed = settings.rpm_min;
          pwm_value = SPINDLE_PWM_MIN_VALUE;
        }
```

The alternative is error, stop job, require reset.

owenduffy, 2019-10-12:
Ok, I understand and accept.

I am about to do a cal run to feed the py script for piecewise linearisation. I am running it on the devt version, and a dry run looks good. The optical tacho will log the actual spindle speed (unloaded).

bdring, 2019-10-12:
Great.

Do you think a SPINDLE_RPM_RANGE_ERROR should be added?

owenduffy, 2019-10-12:
I don't understand.

The cal run looked good with the devt branch.
[image]

owenduffy, 2019-10-12:
I have been going over this, and I think there remains an issue that when the spindle should be stopped, there is a very short (790ns) down pulse in each PWM period.

I have found that if I set sxxx>0 and <rpmmin, I get pwm like rpmmin.

If I set s0, I get no downwards transistions.

If I then hit m2, I see the 790ns down pulse

This seems to be caused by the line commented  by me:

//			grbl_analogWrite(SPINDLE_PWM_CHANNEL, (1<<SPINDLE_PWM_BIT_PRECISION) - 1);
			grbl_analogWrite(SPINDLE_PWM_CHANNEL, (1<<SPINDLE_PWM_BIT_PRECISION));

The problem goes away with this revised line.

bdring, 2019-10-12:
I saw that issue too.

owenduffy, 2019-10-12:
You have partially implemented the piecewise compensation.

I added the code into spindle_control.cpp:

```text
@@ -61,11 +61,12 @@ void spindle_stop()
   spindle_set_enable(false);
 	#ifdef SPINDLE_PWM_PIN
 		#ifndef INVERT_SPINDLE_PWM
 			grbl_analogWrite(SPINDLE_PWM_CHANNEL, SPINDLE_PWM_OFF_VALUE);
 		#else
-			grbl_analogWrite(SPINDLE_PWM_CHANNEL, (1<<SPINDLE_PWM_BIT_PRECISION) - 1);
+//			grbl_analogWrite(SPINDLE_PWM_CHANNEL, (1<<SPINDLE_PWM_BIT_PRECISION) - 1);
+			grbl_analogWrite(SPINDLE_PWM_CHANNEL, (1<<SPINDLE_PWM_BIT_PRECISION));
 		#endif
 	#endif
 }
 
 uint8_t spindle_get_state()  // returns SPINDLE_STATE_DISABLE, SPINDLE_STATE_CW or SPINDLE_STATE_CCW
@@ -138,15 +139,31 @@ uint32_t spindle_compute_pwm_value(float rpm)
 			pwm_value = SPINDLE_PWM_MIN_VALUE;
 			}
 		} else {
 			// Compute intermediate PWM value with linear spindle speed model.
 			// NOTE: A nonlinear model could be installed here, if required, but keep it VERY light-weight.
-			sys.spindle_speed = rpm;
-			pwm_value = floor((rpm-settings.rpm_min)*pwm_gradient) + SPINDLE_PWM_MIN_VALUE;
-		}		
-		
-		return(pwm_value);
+        #if (N_PIECES > 3)
+          if (rpm > RPM_POINT34) {
+            pwm_value = floor(RPM_LINE_A4*rpm - RPM_LINE_B4);
+          } else 
+        #endif
+        #if (N_PIECES > 2)
+          if (rpm > RPM_POINT23) {
+            pwm_value = floor(RPM_LINE_A3*rpm - RPM_LINE_B3);
+          } else 
+        #endif
+        #if (N_PIECES > 1)
+          if (rpm > RPM_POINT12) {
+            pwm_value = floor(RPM_LINE_A2*rpm - RPM_LINE_B2);
+          } else 
+        #endif
+        {
+          pwm_value = floor(RPM_LINE_A1*rpm - RPM_LINE_B1);
+        }
+      }
+   sys.spindle_speed = rpm;
+   return(pwm_value);
 	#else
 		return(0); // no SPINDLE_PWM_PIN
 	#endif
 }
 

```

and cpu_map:

```text
@@ -126,12 +126,14 @@
 	// round down to nearest bit count for SPINDLE_PWM_MAX_VALUE = 13bits (8192)
 	
 	#define SPINDLE_PWM_BASE_FREQ 5000 // Hz
 	#define SPINDLE_PWM_BIT_PRECISION 8   
 	#define SPINDLE_PWM_OFF_VALUE     0
-	#define SPINDLE_PWM_MAX_VALUE     255
-	//#define INVERT_SPINDLE_PWM
+//	#define SPINDLE_PWM_MAX_VALUE     255
+#define INVERT_SPINDLE_PWM
+#define SPINDLE_PWM_MIN_VALUE 3
+#define SPINDLE_PWM_MAX_VALUE 195
 	
 	#ifndef SPINDLE_PWM_MIN_VALUE
 		#define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
 	#endif
 ```

and config.h:

```text
@@ -45,11 +45,12 @@ Some features should not be changed. See notes below.
 // Define CPU pin map and default settings.
 // NOTE: OEMs can avoid the need to maintain/update the defaults.h and cpu_map.h files and use only
 // one configuration file by placing their specific defaults and pin map at the bottom of this file.
 // If doing so, simply comment out these two defines and see instructions below.
 #define DEFAULTS_GENERIC
-#define CPU_MAP_TEST_DRIVE // these are defined in cpu_map.h
+//#define CPU_MAP_TEST_DRIVE // these are defined in cpu_map.h
+#define CPU_MAP_ESP32 // these are defined in cpu_map.h
 #define VERBOSE_HELP // adds addition help info, but could confuse some senders
 
 
 // Serial baud rate
 #define BAUD_RATE 115200
@@ -687,30 +688,27 @@ Some features should not be changed. See notes below.
 #define DISABLE_LASER_DURING_HOLD // Default enabled. Comment to disable.
 
 // Enables a piecewise linear model of the spindle PWM/speed output. Requires a solution by the
 // 'fit_nonlinear_spindle.py' script in the /doc/script folder of the repo. See file comments 
 // on how to gather spindle data and run the script to generate a solution.
-// #define ENABLE_PIECEWISE_LINEAR_SPINDLE  // Default disabled. Uncomment to enable.
+#define ENABLE_PIECEWISE_LINEAR_SPINDLE  // Default disabled. Uncomment to enable.
 
 // N_PIECES, RPM_MAX, RPM_MIN, RPM_POINTxx, and RPM_LINE_XX constants are all set and given by
 // the 'fit_nonlinear_spindle.py' script solution. Used only when ENABLE_PIECEWISE_LINEAR_SPINDLE
 // is enabled. Make sure the constant values are exactly the same as the script solution.
 // NOTE: When N_PIECES < 4, unused RPM_LINE and RPM_POINT defines are not required and omitted.
-#define N_PIECES 4  // Integer (1-4). Number of piecewise lines used in script solution.
-#define RPM_MAX  11686.4  // Max RPM of model. $30 > RPM_MAX will be limited to RPM_MAX.
-#define RPM_MIN  202.5    // Min RPM of model. $31 < RPM_MIN will be limited to RPM_MIN.
-#define RPM_POINT12  6145.4  // Used N_PIECES >=2. Junction point between lines 1 and 2.
-#define RPM_POINT23  9627.8  // Used N_PIECES >=3. Junction point between lines 2 and 3.
-#define RPM_POINT34  10813.9 // Used N_PIECES = 4. Junction point between lines 3 and 4.
-#define RPM_LINE_A1  3.197101e-03  // Used N_PIECES >=1. A and B constants of line 1.
-#define RPM_LINE_B1  -3.526076e-1
-#define RPM_LINE_A2  1.722950e-2   // Used N_PIECES >=2. A and B constants of line 2.
-#define RPM_LINE_B2  8.588176e+01
-#define RPM_LINE_A3  5.901518e-02  // Used N_PIECES >=3. A and B constants of line 3.
-#define RPM_LINE_B3  4.881851e+02
-#define RPM_LINE_A4  1.203413e-01  // Used N_PIECES = 4. A and B constants of line 4.
-#define RPM_LINE_B4  1.151360e+03
+#define N_PIECES 3
+#define RPM_MAX 23935.2
+#define RPM_MIN 2412.2
+#define RPM_POINT12 6283.9
+#define RPM_POINT23 11866.0
+#define RPM_LINE_A1 4.390865e-03
+#define RPM_LINE_B1 7.591787e+00
+#define RPM_LINE_A2 1.074874e-02
+#define RPM_LINE_B2 4.754411e+01
+#define RPM_LINE_A3 9.528342e-03
+#define RPM_LINE_B3 3.306286e+01
 
 
 /* ---------------------------------------------------------------------------------------
    OEM Single File Configuration Option
 

```

This is with cal factors based on my cal run yesterday.

It works on a bench test observing with the scope.

I should be able to run on the machine this morning (weather permitting).

If that is good, it wraps up #244 and #247.

owenduffy, 2019-10-12:
I played with the model fit a bit (changing parameters in the py file), and got a better fit.

This test is with the devt branch with the changes shown earlier to fit the rest of the spindle compensation... and importantly, with inverted spindle PWM.

At the outset:
[image]

After compensation:

[image]

Bart, it works a treat... if you incorporate the rest of the spindle compensation, you can close this and #247.

Thanks for your work... Owen.

bdring, 2019-10-12:
I don't see where #define ENABLE_PIECEWISE_LINEAR_SPINDLE is used in your code.

That code is a lot of work to read in that +/- format.

bdring, 2019-10-12:
OK...looking at original Grbl. It surrounds everything to make an alternative version of the function

owenduffy, 2019-10-12:
It was enabled elsewhere in config.h (just above the stuff I quoted). Shouldn't work without it!!!

Essentially, I anticipated the piecewise code you would add into spindle_control, copied literally from grbl.

I expect you to copy from grbl rather than my code above. I showed that to show the extent of what has been tested.

Happy to retest when you merge the new stuff into devt.

owenduffy, 2019-10-12:
You are thinking I should have forced this and made the changes... but as it is, you are the author and you are more familiar with the code base, I am happy to do the testing.

bdring, 2019-10-12:
I pushed more changes to devt. It is not easy for me to test. 

I made the linear fit a separate function to clean things up and it could allow easier ways to add other fit methods in the future.

owenduffy, 2019-10-13:
Happy to test.

Ok, looks good on a test with the scope (which reveals more than running the machine).

I think that is good to promote.

Thanks again... Owen.

bdring, 2019-10-13:
[Merged](https://github.com/bdring/Grbl_Esp32/pull/251)

closed 2019-10-13 by bdring