bdring/Grbl_Esp32 #247: Provide for conditional include of user spindle curve

owenduffy, 2019-10-09:
**Please describe the feature you would like implemented**

Provide for conditional inclusion of user source file spindle_curve.h to recalculate rpm to linearise response.

**Why do you think this would improve Grbl_ESP32?**
Allows user mod without touching a managed source module

**Will this feature appear to a lot of users?**
Probably.

====
diff --git a/Grbl_Esp32/spindle_control.cpp b/Grbl_Esp32/spindle_control.cpp
index 1886983..4b36600 100644
--- a/Grbl_Esp32/spindle_control.cpp
+++ b/Grbl_Esp32/spindle_control.cpp
@@ -111,6 +111,11 @@ uint32_t spindle_compute_pwm_value(float rpm)
     // Compute intermediate PWM value with linear spindle speed model.
     // NOTE: A nonlinear model could be installed here, if required, but keep it VERY light-weight.
     sys.spindle_speed = rpm;
+#if defined __has_include
+  #if __has_include("spindle_curve.h")
+    #include "spindle_curve.h"
+  #endif
+#endif
     pwm_value = floor((rpm-settings.rpm_min)*pwm_gradient) + SPINDLE_PWM_MIN_VALUE;
   }
   return(pwm_value);
@@ -169,4 +174,3 @@ void spindle_set_enable(bool enable)
 		#endif
 	#endif
 }
-
====

Sample user spindle_curve.h (not to be included in managed source).

====
//optional user customisation of spindle PWM curve
//needs consistent settings.rpm_min and settings.rpm_max
//use block to make vars private
    {
//    static long spdcuri={settings.rpm_min,settings.rpm_max,1000000}
//    static float spdcurm={0,1,0}
//    static long spdcurb={0,0,settings.rpm_max}
    static long spdcuri[]={1200,3960,6720,24000,1000000};
    static float spdcurm[]={0,0.1228,0.5015,0.9074,0};
    static float spdcurb[]={0,1151,-405,-3120,24000};
    int i;
    for(i=0;i<sizeof(spdcuri);i++){
      if(rpm<spdcuri[i]){
        rpm=spdcurm[i]*rpm+spdcurb[i];
        break;
        }
      }
    }
====

Tested and working.

Owen

owenduffy, 2019-10-09:
I might explain that the reason the file is named .h is to work around the way Arduino IDE builds stuff.

owenduffy, 2019-10-11:
A little blog article expanding on an implementation: [CNC6040 router project – spindle speed linearisation](https://owenduffy.net/blog/?p=15786).

bdring, 2019-10-11:
I'll give this a look next week. I think I would reformat it into a more Grbl like style using #defines and cpp files. I will look at implementing the `#define ENABLE_PIECEWISE_LINEAR_SPINDLE` as well.

bdring, 2019-10-13:
[Merged](https://github.com/bdring/Grbl_Esp32/pull/251)

closed 2019-10-13 by bdring