bdring/Grbl_Esp32 #249: Stop work

semos2k, 2019-10-11:
Latest firmware version.
The problem is constant.
I used the zebra milling test on my laser diode CNC the programs tested bCNC(Telnet), laserGrbl(telnet),UGS(Windows Virtual Port - very slow) the cnc stops on the line 9502.

https://drive.google.com/file/d/1_zLdfeZSLuprQjtclKEfZDW82WkD_uqC/view?usp=sharing

semos2k, 2019-10-11:
[image]

bdring, 2019-10-11:
Have you tried running the same file via SD card or USB/Serial port? That would help me debug.

semos2k, 2019-10-11:
From USB work fine.

rafik84, 2019-10-30:
I spent several hours today on something similar, I wrote my own program in delphi for communication via telnet. I send gcode lines every 15ms (about 65 lines per second) (BLOCK_BUFFER_SIZE = 1024 with a few code changes uint8_t! = Uint16_t). Communication with esp was different from a few minutes to an hour and ... ping is esp and communication is gone, responses to usb / uart were returned after about 10s. and what turned out:
in WiFiClient.cpp (wifi / src) it was enough to change lines:
with
#define WIFI_CLIENT_MAX_WRITE_RETRY (10)
#define WIFI_CLIENT_SELECT_TIMEOUT_US (1,000,000)

on

#define WIFI_CLIENT_MAX_WRITE_RETRY (1)
#define WIFI_CLIENT_SELECT_TIMEOUT_US (100000)

everything is great and I can send gcode: D commands even faster

my soft is looks like this :
[IMG]https://images92.fotosik.pl/274/a117771b1f296768med.png[/IMG]
im added 
2x MCP23017 (16 inputs , 16 outputs OC)
1x485 to controll VFD

luc-github, 2019-10-30:
@rafik84 you send faster no doubt, so decreasing number of retries/timeout will decrease transfert time but what about data integrity ? Because retry is happening if issue occur, ignoring it may have consquence. 

Question : Do you do data integrity check to verify everything is received and same as sent?

rafik84, 2019-10-30:
Yes , im control line gcode by N0000 .. X  added to send line and everiting is fine ,decreasing number of retries it's  posible  lost data recv from ESP but .Im connecting to esp as AP then i'm have ping between 1ms < 3ms ,im using 2 telnet server / ports ,one to send gcode line and recv OK from parser (and this is work fine) and 2 to recv/send commands system (report,ovr) to separate layers system/gcode parser ,in report im always control state of buffer planner with do not cross < 100 when im lose response from report_real