bdring/Grbl_Esp32 #250: Homing doesnt work as intended

karoria, 2019-10-11:
I use NO type inductive proximity limit switches for my mini vmc machine. Somehow, I am not able to configure homing correctly. The switches responds as hard limits but for homing it has to rapidly sense object two times to successfully home any axis. If i take more than a second in sensed position, message of homing failed pops up. I have tried almost all the combination of homing related parameters but no luck. Please guide me where I might have missed something.

bdring, 2019-10-11:
I don't understand.

Move the X axis far away from the switch. 
  - Do you see a switch activated with the ? command.
 - Send $HX. Does the axis move towards the switch?
 - Does is reverse when it gets close to the switch?
 - Does it go back for the second attempt or fail?

Have you tried adjusting the $27 "Homing switch pull-off distance" to a larger value?

JackHouseLab, 2019-10-11:
Hi,
Try to set $5 "Limit pins invert, boolean" to 1 - it helped for me.

karoria, 2019-10-12:
Thanks for prompt response @bdring and @JackHouseLab.
I have already tried $5 invert but no luck.
Will try the tests Barton has suggested and will reply soon.

karoria, 2019-10-16:
Sorry to bother you for a silly mistake.!! Actually the switch was not disengaging properly as $27 homing switch pull off distance was only 2mm. Increased it and solved. 
Thanks again for your valuable time.

closed 2019-10-16 by karoria

ggallant571, 2019-10-16:
Not a silly mistake, valuable lesson. I never would have caught it!!!