bdring/Grbl_Esp32 #252: Error in spindle_control.cpp, spindle_set_speed()

owenduffy, 2019-10-15:
Please answer the following questions.

What version of the firmware are you using?
b7ee4265515757c82a732c4b19e3a976ffc39a56

Is the problem repeatable?
Yes

Under what conditions does the bug occur?
Using INVERT_SPINDLE_PWM, there is an intercept error in the characteristic of measured PWM vs requested PWM.
[image]
The intercept should be zero.

The following changes fixes the problem, results in intercept of 2e-5 which is good.

```text
@@ -108,11 +108,12 @@ void spindle_set_speed(uint32_t pwm_value)
 		#else
 			if (pwm_value == 0) {
 				grbl_analogWrite(SPINDLE_PWM_CHANNEL, (1<<SPINDLE_PWM_BIT_PRECISION));
 			}
 			else {
-				grbl_analogWrite(SPINDLE_PWM_CHANNEL, (1<<SPINDLE_PWM_BIT_PRECISION) - pwm_value - 1);
+//od				grbl_analogWrite(SPINDLE_PWM_CHANNEL, (1<<SPINDLE_PWM_BIT_PRECISION) - pwm_value - 1);
+				grbl_analogWrite(SPINDLE_PWM_CHANNEL, (1<<SPINDLE_PWM_BIT_PRECISION) - pwm_value);
 			}			
 		#endif
 		
 	#endif
```

bdring, 2019-10-15:
Is this correct? I have trouble reading your code formatting.

```C++
void spindle_set_speed(uint32_t pwm_value)
{	
	#ifndef SPINDLE_PWM_PIN
		return;
	#else
		#ifndef SPINDLE_ENABLE_OFF_WITH_ZERO_SPEED
			spindle_set_enable(true);
		#else
			spindle_set_enable(pwm_value != 0);
		#endif
		
		#ifndef INVERT_SPINDLE_PWM
			grbl_analogWrite(SPINDLE_PWM_CHANNEL, pwm_value);			
		#else
			if (pwm_value == 0) {
				grbl_analogWrite(SPINDLE_PWM_CHANNEL, (1<<SPINDLE_PWM_BIT_PRECISION));
			}
			else {
				grbl_analogWrite(SPINDLE_PWM_CHANNEL, (1<<SPINDLE_PWM_BIT_PRECISION) - pwm_value);
			}			
		#endif
		
	#endif
	
}
```

owenduffy, 2019-10-15:
YEs, but it simplifies to:

```text
void spindle_set_speed(uint32_t pwm_value)
{	
	#ifndef SPINDLE_PWM_PIN
		return;
	#else
		#ifndef SPINDLE_ENABLE_OFF_WITH_ZERO_SPEED
			spindle_set_enable(true);
		#else
			spindle_set_enable(pwm_value != 0);
		#endif
		
		#ifndef INVERT_SPINDLE_PWM
			grbl_analogWrite(SPINDLE_PWM_CHANNEL, pwm_value);			
		#else
			grbl_analogWrite(SPINDLE_PWM_CHANNEL, (1<<SPINDLE_PWM_BIT_PRECISION) - pwm_value);
		#endif
		
	#endif
	
}
```

Ok?

bdring, 2019-10-15:
OK

owenduffy, 2019-10-15:
Apologies for not finding that in testing... I attributed the small error to the piecewise approximation. Fixing this improves cal at the low end.

Thanks again.

bdring, 2019-10-15:
merged to master

closed 2019-10-15 by bdring