bdring/Grbl_Esp32 #257: Option to have Laser mode disable the Spindle

antibyte, 2019-10-19:
It would be nice to have a compile time option to disable the spindle if the laser mode is enabled.
I think this should be fairly easy to add.
I made a PCB for my mini CNC and have spindle enable connected to a mosfet driver so it can shutdown the spindle even if PWM pulses are coming in. The PWM line also goes to the laser driver, but i have to disconnect the spindle in laser mode as it keeps beeing enabled. With such an option i could have the spindle connected and parked in an holder when switching to a laser module.

bdring, 2019-10-19:
Could you attach a PDF of your proposed PCB? It would help understand your application.

antibyte, 2019-10-20:
https://easyeda.com/andreasjakob/esp32grbl_copy

bdring, 2019-10-21:
Do you want to still use spindle enable with the laser?  Could that be used to decide which one gets the PWM?

antibyte, 2019-10-22:
If Laser mode is on ($32=1) spindle enable should be off, so the laser gets its pwm but the motor will be disabled.

bdring, 2019-10-22:
OK, I'll add that to the road map. I am rather busy right now, so it may take a few weeks.

maximius, 2020-04-07:
Good afternoon. Is it planned to implement the function of disabling the relay output when turning on laser mode $32=1