bdring/Grbl_Esp32 #259: Autolevel

hectorbarranca, 2019-10-22:
Recently I started to check this project, and sounds wow, super great!!

Actually I use my CNC to create PCB boards, the problem is that sometimes the boards level is not correct so I use an autolevel software, named bCNC also in github:
https://github.com/vlachoudis/bCNC

The request is about implementing somethink similar, the actual process is like:
-	In the software of bCNC, the user put the Z start position and Z end position of the prove (the end position only for security), the number of points to check for X, Y and the margin but you can put “automargin” that check in the file the minimum and maximum X and Y position to obtain the size of the board
-	The software starts to check the level of Z (you need to put the prove to close the circuit with the conductivity of the drill and the pcb)
-	(not sure) Create a temporary level file and make a second temporary file with the original file and corrections

I think the first step can be improved to reduce the work to the user, for example you can only ask the distance between prove check (for example 5mm) and automatic automargin. For example if the board lengt is 13mm and it is pre-configured to 5mm the number of point can be ceil(13/5) + 1 = 4, and the points in 0, 3.25, 6.5, 9.75, 13

Thanks for your time to check the idea, I am not sure if I can help but I can try =).

bdring, 2019-10-22:
Auto level is not something I want to add to Grbl_ESP32 at this time.  I think a sender with a good user interface is a much better place for it.

Also, Grbl does not know anything about the gcode file before it runs. It does not know the size of your material or even the extents of the moves.

hectorbarranca, 2019-10-22:
Thanks my friend, no problem =)

closed 2019-10-22 by bdring

hitronics4trade, 2019-10-23:
Sir, bdring
I have a question relating auto level with your control board :
is it auto level for one point or auto level for multi points  
I asked this question because sometimes the surface maybe curve slightly