bdring/Grbl_Esp32 #261: Everytime I connect the board, I have to hit reset button

karoria, 2019-10-23:
I dont know whether it is a problem or not, but each time I connect the board to software, I need to hit reset to enable all controls of the software. Any workaround to get rid of it?

mac7988, 2019-10-23:
Hi Ravi,

This might be a blessing. There might unfavorable condition that you are not aware of. So a alarm reset and homing is my way to protect the machine and myself from idiot move.

bdring, 2019-10-23:
The arduino bootloaders typically reboot on connect. Some senders rely on this. ESP32 does not. This is a good thing.

Many modern grbl senders send a soft reset or don't care. What sender are you using?

karoria, 2019-10-23:
Good if it is purposely and helpful. I use CNCjs and just after connecting the board, I have to give reset (ctrl+x) and then console shows eeprom settings etc.

karoria, 2019-10-23:
I will keep it as is. Just for curiosity, Is there any process to eliminate this reset process at boot? I have also noted that this happens when board gets power on. If you dont power off and simply disconnect the com port, then it doesn't require reset.

bdring, 2019-10-23:
It is not an ESP32 firmware issue, it is a hardware issue or gcode sender issue. It **cannot** be done via firmware with the ESP32 modules.

Rebooting Grbl_ESP32 is not desirable. Unlike arduino Grbl, The ESP32 might have active Wifi, Bluetooth or SD jobs running. Connecting via serial should not kill those connections and destroy your running job.

Please have CNCJS solve this. Many senders send a soft reset or some other Grbl command to get what they need when they connect.

karoria, 2019-10-24:
Got it. Thanks.

closed 2019-10-24 by karoria