bdring/Grbl_Esp32 #262: Relating auto level with your control board

hitronics4trade, 2019-10-23:
Sir, bdring
I have a question relating auto level with your control board :
is it auto level for one point or auto level for multi points
I asked this question because sometimes the surface maybe curve slightly

bdring, 2019-10-23:
There is no "autolevel". There is probing of a single point. Probing is supported in the WebUI and many gcode senders.

Some senders will probe a grid and modify the gcode to compensate for an uneven surface. bCNC is a good one.

hitronics4trade, 2019-10-23:
so how to make grid probing?

bdring, 2019-10-23:
Search Youtube for "bCNC Autolevel". [Like this one](https://www.youtube.com/watch?v=3yNVt_2uwwM) 

bCNC has a very old school interface, but the features are great!

closed 2019-10-24 by bdring

PolSerg, 2020-10-26:
Maybe you know an analogue for android?