bdring/Grbl_Esp32 #266: During the movement of any axis, the ECS spindle stops.

hedon77, 2019-10-25:
What version of the firmware are you using?
1.1f (20191023) (ESP32)

Is the problem repeatable?
Yes.

Under what conditions does the bug occur?
During the movement of any axis, the ECS spindle stops.
The ECS controller is still issuing 2 beeps.
Everything worked in version 1.1f (20190708) (ESP32).

Best regards

bdring, 2019-10-25:
I'll look into that tomorrow. Could you paste in your cpu_map (just the text of the one you use) and your $$ settings?

hedon77, 2019-10-26:
Hi,

This is corexy system.

#ifdef CPU_MAP_ESP32_ESC_SPINDLE
	// This is the CPU Map for the ESP32 CNC Controller R2	
	
	  // It is OK to comment out any step and direction pins. This
    // won't affect operation except that there will be no output
		// form the pins. Grbl will virtually move the axis. This could 
		// be handy if you are using a servo, etc. for another axis.
		#define CPU_MAP_NAME "CPU_MAP_ESP32_ESC_SPINDLE"	
		
		#define X_STEP_PIN      	GPIO_NUM_12
		#define X_DIRECTION_PIN  	GPIO_NUM_26
		#define X_RMT_CHANNEL		0
		
		#define Y_STEP_PIN      	GPIO_NUM_14
		#define Y_DIRECTION_PIN   	GPIO_NUM_25
		#define Y_RMT_CHANNEL		1
		
		#define Z_STEP_PIN      	GPIO_NUM_27		
		#define Z_DIRECTION_PIN   	GPIO_NUM_33 		 
		#define Z_RMT_CHANNEL		2		
		
		// OK to comment out to use pin for other features
		#define STEPPERS_DISABLE_PIN GPIO_NUM_13		
		
		// *** the flood coolant feature code is activated by defining this pins
		// *** Comment it out to use the pin for other features
		#define COOLANT_FLOOD_PIN 	GPIO_NUM_16			
		//#define COOLANT_MIST_PIN   	GPIO_NUM_21
		
		// If SPINDLE_PWM_PIN is commented out, this frees up the pin, but Grbl will still
		// use a virtual spindle. Do not comment out the other parameters for the spindle.
		#define SPINDLE_PWM_PIN    GPIO_NUM_17 
		#define SPINDLE_PWM_CHANNEL 0
		
		// RC ESC Based Spindle 
		// An ESC works like a hobby servo with 50Hz PWM 1ms to 2 ms pulse range		
		#define SPINDLE_PWM_BASE_FREQ 50 // Hz for ESC
		#define SPINDLE_PWM_BIT_PRECISION 16   // 16 bit required for ESC
		#define SPINDLE_PULSE_RES_COUNT 65535
		
		#define ESC_MIN_PULSE_SEC 0.001 // min pulse in seconds (OK to tune this one)
		#define ESC_MAX_PULSE_SEC 0.002 // max pulse in seconds (OK to tune this one)		
		#define ESC_TIME_PER_BIT  ((1.0 / (float)SPINDLE_PWM_BASE_FREQ) / ((float)SPINDLE_PULSE_RES_COUNT) ) // seconds

		#define SPINDLE_PWM_OFF_VALUE    (uint16_t)(ESC_MIN_PULSE_SEC / ESC_TIME_PER_BIT) // in timer counts
		#define SPINDLE_PWM_MAX_VALUE    (uint16_t)(ESC_MAX_PULSE_SEC / ESC_TIME_PER_BIT) // in timer counts
		
		#ifndef SPINDLE_PWM_MIN_VALUE
				#define SPINDLE_PWM_MIN_VALUE   SPINDLE_PWM_OFF_VALUE   // Must be greater than zero.
		#endif
				
		#define SPINDLE_ENABLE_PIN	GPIO_NUM_22			
		
		// if these spindle function pins are defined, they will be activated in the code
		// comment them out to use the pins for other functions
		//#define SPINDLE_ENABLE_PIN	GPIO_NUM_16
		//#define SPINDLE_DIR_PIN			GPIO_NUM_16		
		
		#define X_LIMIT_PIN      	GPIO_NUM_35 
		#define Y_LIMIT_PIN      	GPIO_NUM_4  
		#define Z_LIMIT_PIN     	GPIO_NUM_15 	
		#define LIMIT_MASK      	B111
		
		#define PROBE_PIN       	GPIO_NUM_32  
		
		//#define CONTROL_SAFETY_DOOR_PIN   GPIO_NUM_35  // needs external pullup
		#define CONTROL_RESET_PIN         GPIO_NUM_34  // needs external pullup
		#define CONTROL_FEED_HOLD_PIN     GPIO_NUM_36  // needs external pullup 
		#define CONTROL_CYCLE_START_PIN   GPIO_NUM_39  // needs external pullup    		
		
#endif	

$$
$0=3
$1=250
$2=0
$3=0
$4=0
$5=1
$6=0
$10=15
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=4
$24=2000.000
$25=100.000
$26=250
$27=3.000
$30=1000.000
$31=0.000
$32=0
$100=100.000
$101=100.000
$102=400.000
$110=2000.000
$111=2000.000
$112=400.000
$120=200.000
$121=200.000
$122=200.000
$130=300.000
$131=300.000
$132=300.000

Best regards

hedon77, 2019-10-26:
Hi,

My schemat. [https://cloud.24box.pl/index.php/s/miGBAwqJxYicC4W](url)

bdring, 2019-10-26:
@hedon77 

I think I may have found the problem. Original Grbl only supports 8 bit PWM. 16 bit PWM is needed to accurately run Servo/ESC type PWMs.  The stepper module was storing and resetting the PWM using an 8 bit value, so it was corrupting the number.

I need to do some more testing and cleanup the debugging code I added to make sure.

I don't know when/how the problem got in. My ESC test rig takes a long time to setup, so I don't use it very often.

bdring, 2019-10-26:
@hedon77 

I pushed some fixes to the devt branch. Can you try them? I made some changes to the "CPU_MAP_ESP32_ESC_SPINDLE" cpu map. If you were using that, you will need make a new one.

There is also a [wiki page for the feature now](https://github.com/bdring/Grbl_Esp32/wiki/BESC-Spindle-Feature).

I noticed your schematic had "ANT" in the title. Is this for an ANT PCB Maker. I would love to see some photos when you get it running.

hedon77, 2019-10-27:
Hi,

After uploading the dev repository, the Y axis motor stopped working. The spindle also does not work.

This is how I submit an "Ant", as soon as I finish sharing a film about the action: D

best regards

bdring, 2019-10-27:
I changed the cpu map. Could that be the problem?

hedon77, 2019-10-30:
303/5000
After mapping the pins, the axes work. The spindle works but still makes a sound. Is this splicing procedure necessary? Earlier operation of the spindle was simpler and more convenient for me, maybe you could do this new procedure optionally?
On the weekend I will write a video from the machine's operation.

bdring, 2019-10-30:
What splicing?  What sound? What procedure?

hedon77, 2019-10-30:
What splicing? 

Turn on sequence necessary?

What sound? 

The film will show, the spindle after turn on sequence makes constant sound.

What procedure?

Turn on sequence.

Sorry for my english this is not my native language.

thx best regards

closed 2019-11-11 by bdring