bdring/Grbl_Esp32 #268: What pins are the limit switches

jaysettle, 2019-10-26:
Sorry to post this as an issue, I'm looking in all the wrong places but where is the pinout for the limit switches? What circuit do you suggest to help with noise? Looks like some resistors present here.
[image]

_These also have R/C filters to eliminate high frequency noise from false triggering the switches._

Also where is the entire pinout mapping?

I looked through the following:
https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins
https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/grbl_limits.cpp
https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/grbl_limits.h
https://github.com/bdring/Grbl_Esp32
https://github.com/bdring/Grbl_Esp32/wiki/Controlling-Grbl_ESP32
http://www.buildlog.net/blog/wp-content/uploads/2018/10/outline_esp32_cnc_test_v3p1.pdf

jaysettle, 2019-10-26:
Ok I think I found them here in the pdf. Is there an easier path for people to find the pin mapping? Thanks for your work here!
http://www.buildlog.net/blog/wp-content/uploads/2019/05/schm_esp32_cnc_test_v3p5.pdf

[image]

And so you appear to use an RC filter. What values would you recommend? NO or NC on the switch?
[image]

derFrickler, 2019-10-26:
Just look them up or change them in the CPU map:
https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/cpu_map.h#L92

bdring, 2019-10-26:
The cpu map is this one

```C++
#ifdef CPU_MAP_ESP32
	// This is the CPU Map for the ESP32 Development Controller
	// https://github.com/bdring/Grbl_ESP32_Development_Controller
	// https://www.tindie.com/products/33366583/grbl_esp32-cnc-development-board-v35/
        ...
```
I suggest using the V4 version.

There is also limit switch debouncing that can help with noise.

jaysettle, 2019-10-26:
Why V4? I have V3 working, what's the difference? Thanks.

bdring, 2019-10-26:
Some i/o was shuffled around on V4. The primary reason was io2. Most dev boards have an LED on io2 and that can give problems if you use it as an input. For some reason, I have never had the problem, but many people have reported it. It must be close to a threshold.

io2 is used for the x limit switch on V3 and earlier. On V4 it is the spindle PWM. It works well as a PWM output because you can see the brightness change.  A few other things shuffled around too.

[Everything I know about ESP32 gpio is on this wiki page](https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins). I add to it every time I learn something new.

TimeWaster4000, 2019-11-04:
Why is reset tied to P34 when the esp32 dev kit has the reset on the EN pin.
I would like to use P34 for end stop but when you use a pullup and the end stop is triggered it resets the unit.
I have commented out   ( //#define CONTROL_RESET_PIN         GPIO_NUM_34 ) still resets the unit
Thank You

bdring, 2019-11-04:
It is a different type of reset. Grbl reset is a way to "soft" reset grbl.  

 - If Grbl is using hard or soft limit switches, you need do a soft reset to clear the error
 - If you are running a job and do a feed hold, you can either restart the job or do a Grbl reset to end it.

Generally a soft reset clears out Grbl's buffer, but you do not loose position and don't need to rehome. 

You don't need a button. The soft reset can also be done with a Gcode sender.

Attach your cpu_map

TimeWaster4000, 2019-11-04:
Here is my cpu_map and read me txt.
[lpcnc.zip](https://github.com/bdring/Grbl_Esp32/files/3806497/lpcnc.zip)

bdring, 2019-11-04:
Please paste your map into the reply.

...just the one you are using.

TimeWaster4000, 2019-11-04:
#ifdef CPU_MAP_ESP32_LPCNC
  // This is the CPU Map for the LPCNC Shield   
  
  // It is OK to comment out any step and direction pins. This
    // won't affect operation except that there will be no output
    // form the pins. Grbl will virtually move the axis. This could 
    // be handy if you are using a servo, etc. for another axis.
    #define CPU_MAP_NAME "CPU_MAP_ESP32_LPCNC"  
    
    #define X_STEP_PIN        GPIO_NUM_12
    #define X_DIRECTION_PIN   GPIO_NUM_14
    #define X_RMT_CHANNEL   0
    
    #define Y_STEP_PIN        GPIO_NUM_27
    #define Y_DIRECTION_PIN     GPIO_NUM_26
    #define Y_RMT_CHANNEL   1
    
    #define Z_STEP_PIN        GPIO_NUM_25   
    #define Z_DIRECTION_PIN     GPIO_NUM_33      
    #define Z_RMT_CHANNEL   2   
    
    // OK to comment out to use pin for other features
    #define STEPPERS_DISABLE_PIN GPIO_NUM_13    
    
    // *** the flood coolant feature code is activated by defining this pins
    // *** Comment it out to use the pin for other features
    // #define COOLANT_FLOOD_PIN   GPIO_NUM_16     
    //#define COOLANT_MIST_PIN    GPIO_NUM_21
    
    // If SPINDLE_PWM_PIN is commented out, this frees up the pin, but Grbl will still
    // use a virtual spindle. Do not comment out the other parameters for the spindle.
    #define SPINDLE_PWM_PIN    GPIO_NUM_17 
    #define SPINDLE_PWM_CHANNEL 0
    // PWM Generator is based on 80,000,000 Hz counter
    // Therefor the freq determines the resolution
    // 80,000,000 / freq = max resolution
    // For 5000 that is 80,000,000 / 5000 = 16000 
    // round down to nearest bit count for SPINDLE_PWM_MAX_VALUE = 13bits (8192)
    #define SPINDLE_PWM_BASE_FREQ 5333 // Hz
    #define SPINDLE_PWM_BIT_PRECISION 8   // be sure to match this with SPINDLE_PWM_MAX_VALUE
    #define SPINDLE_PWM_OFF_VALUE     0
    #define SPINDLE_PWM_MAX_VALUE     255 // (2^SPINDLE_PWM_BIT_PRECISION)
    
    #ifndef SPINDLE_PWM_MIN_VALUE
        #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
    #endif
    
    // #define SPINDLE_ENABLE_PIN  GPIO_NUM_22
    
    #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)       
    
    // if these spindle function pins are defined, they will be activated in the code
    // comment them out to use the pins for other functions
    //#define SPINDLE_ENABLE_PIN  GPIO_NUM_16
    //#define SPINDLE_DIR_PIN   GPIO_NUM_16   
    
    #define X_LIMIT_PIN       GPIO_NUM_34  
    #define Y_LIMIT_PIN       GPIO_NUM_35  
    #define Z_LIMIT_PIN       GPIO_NUM_32   
    #define LIMIT_MASK        B111
    
    // if using a probe, Door, Reset, Hold, Start, Servo or Solenoid uncomment and asign a pin
    // Useable pins are 0,15, 21, 22, 32, 34, 35, 36, 39
    // Pins 34, 35, 36 and 39 need external pullup
    //#define PROBE_PIN             GPIO_NUM_32  
    //#define CONTROL_SAFETY_DOOR_PIN   GPIO_NUM_35   
    //#define CONTROL_RESET_PIN         GPIO_NUM_34  
    //#define CONTROL_FEED_HOLD_PIN     GPIO_NUM_36   
    //#define CONTROL_CYCLE_START_PIN   GPIO_NUM_39  
    
    //#define USING_SERVO  // uncomment to use this feature
    //#define USING_SOLENOID // uncomment to use this feature
    
    #ifdef USING_SERVO
      #define USE_SERVO_AXES 
      #define SERVO_Z_PIN   GPIO_NUM_27
      #define SERVO_Z_CHANNEL_NUM 3
      #define SERVO_Z_RANGE_MIN 0
      #define SERVO_Z_RANGE_MAX 10
    #endif
        
    #ifdef USING_SOLENOID
      #define USE_PEN_SOLENOID
      #define SOLENOID_PEN_PIN GPIO_NUM_16
      #define SOLENOID_CHANNEL_NUM 6
    #endif
        
    
#endif

This cpu_map entry is for my diy board

bdring, 2019-11-05:
What exactly is the problem you are experiencing?  I just loaded your cpu map onto an ESP32 dev kit and the x limit works fine.

This is cut from my serial monitor. This shows status from the ? command. I connected pin 34 to ground and see pin X and connect it to 3.3v and don't see X.

No resets.

```text
[MSG:Using cpu_map...CPU_MAP_ESP32_LPCNC]

Grbl 1.1f ['$' for help]
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:XY|WCO:0.000,0.000,0.000>
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:XY|Ov:100,100,100>
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:XY>
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:XY>
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:XY>
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:Y>
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:XY>
ok

```

TimeWaster4000, 2019-11-05:
Works fine without pull up when you add a pullup and trigger the limit switch it resets.
I have looked all over the place in the firmware for a reference to P34 to move the grbl_reset to a different pin.
Is the GRBL_Reset hard coded to that pin.

bdring, 2019-11-05:
The pullup should not affect anything. Are you pulling up to 3.3V?

There are no hard coded pins in the firmware. It is all referenced to the cpu map. 

The limit switch operation is defined in system.cpp. They are attached to an interrupt where they are read. System.cpp gets all GPIO_NUMs from cpu map.

bdring, 2019-11-05:
Are you seeing text like **[MSG:Reset via control pin]** in the serial console?

TimeWaster4000, 2019-11-05:
Yes pull 3.3V just checked it to be sore.
Console output when limit triggered.

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Axis count 3]
[MSG:Timed Steps]

[MSG:Client Started]
[MSG:Connecting The_Monster]
[MSG:Connecting.]
[MSG:Connected with 192.168.2.13]
[MSG:Connected with 192.168.2.13]
[MSG:Start mDNS with hostname:http://Laser.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Using cpu_map...CPU_MAP_ESP32_LPCNC]

Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]

bdring, 2019-11-05:
That is hard reset **(POWERON_RESET)**. Grbl reset is not doing that.

Grbl reset via button will send this text to the serial port **[MSG:Reset via control pin]**. You will not see that startup text.

That is not a firmware crash or watchdog timeout either. With a crash you will see some backtrace information.

I get what you are getting by clicking the ESP32 reset button which is connected to the EN pin.

TimeWaster4000, 2019-11-05:
Wow I mist it.
Thank you for your patience.
The problem is nether the firmware or the board.
It is the endstop malfunctioning.
Thank you again I was looking in the wrong place.

bdring, 2019-11-05:
OK, Good.

The serial console will always offer a lot of clues.

closed 2019-11-05 by bdring